openthread/openthread #4075: Run entire NCP thread-cert suite

bstraehl, 2019-08-08:
I try to run the entire NCP thread-cert suite according to [this](https://github.com/openthread/pyspinel/blob/master/README.md#run-entire-ncp-thread-cert-suite).

Information to reproduce the behavior:
1. Git commit 6ebc5174087fbbe507f8132935d4f55ed7f6d1be
2. on posix
3. gcc: gcc (Debian 8.3.0-19) 8.3.0
4. Build steps:
- git clone https://github.com/openthread/openthread.git
- cd openthread
- make distclean
- ./bootstrap
- NODE_TYPE=ncp-sim BUILD_TARGET=posix-distcheck DISTCHECK_CONFIGURE_FLAGS="--with-examples=posix --enable-cli-app --enable-ncp-app=all --with-ncp-bus=uart --with-tests=all" make -f examples/Makefile-posix distcheck BuildJobs=10 VERBOSE=1
5. The build of libopenthread_posix_a fails with 
../../../../../examples/platforms/posix/platform-posix.h:61:10: fatal error: platform-config.h: No such file or directory
 #include "platform-config.h"
          ^~~~~~~~~~~~~~~~~~~
for several files.

6. Adding ```"-I$(top_srcdir)/examples/platforms/posix      \"``` to Makefile.am in ```examples/platforms/posix``` doesn't help, although ```"-I/opt/test/openthread/examples/../examples/platforms/posix/"``` is added to CFLAGS.

Full console log: [openthread.build.log](https://github.com/openthread/openthread/files/3481941/openthread.build.log)

jwhui, 2019-08-08:
@bstraehl, thanks for raising this issue.

The documentation you referenced is out-of-date.

To install python dependencies:
```
$ pip install -r tests/scripts/thread-cert/requirements.txt 
$ pip install git+https://github.com/openthread/pyspinel
```
Then try:
```
$ ./bootstrap
$ NODE_TYPE=ncp-sim make -f examples/Makefile-posix check
```

For future reference, you can take a look at what our Travis scripts do:

- [.travis/before_install.sh](https://github.com/openthread/openthread/blob/f740fda8d66720096846da99615501b7e2de4a73/.travis/before_install.sh#L43)
- [.travis/script.sh](https://github.com/openthread/openthread/blob/f740fda8d66720096846da99615501b7e2de4a73/.travis/script.sh#L444)

Hope that helps.

bstraehl, 2019-08-12:
@jwhui Thank you for your hints. I'm working always with the Debian Testing (currently code name Bullseye) release, which provides quite up-to-date packages. And of course I prefer to install native packages. But now I installed the versions mentioned in tests/scripts/thread-cert/requirements.txt using pip. Now 
NODE_TYPE=ncp-sim make -f examples/Makefile-posix check
works and is green (PASS) up to now.
Thank you.

closed 2019-08-12 by bstraehl