openthread/openthread #4079: [nrf52840] undefined reference to `app_error_handler'

bukepo, 2019-08-09:
**Describe the bug**
undefined reference to `app_error_handler' when build with DEBUG=1

**To Reproduce**
Information to reproduce the behavior, including:
1. make -f examples/Makefile-nrf52840 DEBUG=1 USB=1 

**Expected behavior**
build successfully.

**Console/log output**
../../../third_party/NordicSemiconductor/libnordicsemi-nrf52840-sdk.a(libnordicsemi_nrf52840_sdk_a-app_usbd.o): In function `app_usbd_init':
third_party/NordicSemiconductor/libraries/usb/app_usbd.c:863: undefined reference to `app_error_handler'
../../../third_party/NordicSemiconductor/libnordicsemi-nrf52840-sdk.a(libnordicsemi_nrf52840_sdk_a-app_usbd.o): In function `app_usbd_power_events_enable':
third_party/NordicSemiconductor/libraries/usb/app_usbd.c:962: undefined reference to `app_error_handler'
collect2: error: ld returned 1 exit status

**Additional context**
None

bukepo, 2019-08-19:
@LuDuda any thoughts?

LuDuda, 2019-08-19:
@bukepo Thanks for reporting this issue, I added it to the current sprint, will take a look very soon.

LuDuda, 2019-08-31:
I think we can close this PR thanks to @bukepo.

jwhui, 2019-08-31:
Resolved by #4124

closed 2019-08-31 by jwhui