openthread/openthread #4092: Running single thread-cert tests on SOCs

bstraehl, 2019-08-12:
I try to run single thread certification tests on SOCs. For instance:
```NODE_TYPE=soc python Cert_6_3_01_OrphanReattach.py```
As test systems I took 3 randomly chosen EFR32MG12 based SOCs and flashed them with ot-cli-ftd of git commit id b51ab6c45a1edc180ba48eb2fd94afca18094d57. 
The test didn't pass because the device supposed to be router, always formed its own network. After a short analysis of Cert_6_3_01_OrphanReattach.py, I've seen that the test does not define a master key for the 3 devices (leader, router and ed). So I compared the master keys of my 3 devices, using a terminal program and indeed the keys where different. And as we learned in https://github.com/openthread/openthread/issues/4036, the only important information to form a Thread network is the master key. 
So I added this:
```
diff --git a/openthread/tests/scripts/thread-cert/Cert_6_3_01_OrphanReattach.py b/openthread/tests/scripts/thread-cert/Cert_6_3_01_OrphanReattach.py
index c3a8e661..d70b6e74 100755
--- a/openthread/tests/scripts/thread-cert/Cert_6_3_01_OrphanReattach.py
+++ b/openthread/tests/scripts/thread-cert/Cert_6_3_01_OrphanReattach.py
@@ -46,11 +46,13 @@ class Cert_6_3_1_OrphanReattach(unittest.TestCase):
             self.nodes[i] = node.Node(i, (i == ED), simulator=self.simulator)
 
         self.nodes[LEADER].set_panid(0xface)
+        self.nodes[LEADER].set_masterkey("00010203040506070809101112131415")
         self.nodes[LEADER].set_mode('rsdn')
         self.nodes[LEADER].add_whitelist(self.nodes[ROUTER].get_addr64())
         self.nodes[LEADER].enable_whitelist()
 
         self.nodes[ROUTER].set_panid(0xface)
+        self.nodes[ROUTER].set_masterkey("00010203040506070809101112131415")
         self.nodes[ROUTER].set_mode('rsdn')
         self.nodes[ROUTER].add_whitelist(self.nodes[LEADER].get_addr64())
         self.nodes[ROUTER].add_whitelist(self.nodes[ED].get_addr64())
@@ -58,6 +60,7 @@ class Cert_6_3_1_OrphanReattach(unittest.TestCase):
         self.nodes[ROUTER].set_router_selection_jitter(1)
 
         self.nodes[ED].set_panid(0xface)
+        self.nodes[ED].set_masterkey("00010203040506070809101112131415")
         self.nodes[ED].set_mode('rsn')
         self.nodes[ED].add_whitelist(self.nodes[ROUTER].get_addr64())
         self.nodes[ED].enable_whitelist()
slb@simdev.defence.ruag.com@fwHD83SNML:~/projects/pan/pan/openthread$ 
```
to the test. Now it works perfectly as intended. 
This raises the question: Are the Thread certification tests supposed to be run all in sequence (because there is maybe an earlier one, which defines master keys) or are they meant to be "self-contained"? 
By the way the same test is running properly on Posix. Maybe because the whole active data set or just the master key is memset to 0 and thus is the same for every device involved.

jwhui, 2019-08-12:
@bstraehl, the python tests assume that the device is in a factory-reset state at the start. Could it be that your devices have some OpenThread settings stored in non-volatile?

You are correct that the primary use of the python tests currently is for continuous-integration testing on Travis (i.e. in a simulated environment) rather than on real hardware.

bstraehl, 2019-08-13:
@jwhui , yes my devices had OpenThread settings in the NVM. That's the reason, why the keys were different. So before running any tests, SOCs have to be factory reset. I quickly grepped through all the thread-cert tests and there is no one, which does this. It would be very efficient to have one, which just executes a factory reset on the devices, which could be executed before the real tests. 
But as I understand the main purpose of hte thread-cert tests is to run during continuous integration, which is done only on PCs anyway. 
Thank you.

closed 2019-08-13 by bstraehl