openthread/openthread #4095: CMake Support

MrPointer, 2019-08-13:
Hi, I was wondering whether there are any plans to add cmake integration/support in the near future, or at all?
I feel I don't need to elaborate on why cmake is a good (and IMHO better) option to maintain the project's structure and build process.

bukepo, 2019-08-19:
We have a CMakefile for OpenThread in https://github.com/openthread/ot-rtos/blob/master/third_party/openthread/CMakeLists.txt

I don't think it a bad idea to add native support. Thoughts? @jwhui @abtink @gjc13

MrPointer, 2019-08-19:
@bukepo Hmm, didn't notice that - Never would've thought it'll be under the `third_party` dir.
However it'll be nice to have CMake a 1st-grade citizen here.

jwhui, 2019-08-19:
@MrPointer, thank you for submitting this issue.

I think it's worth considering CMake support. I just submitted #4112 to help drive the discussion on what that would look like.

jwhui, 2019-11-25:
Resolved by #4112.

closed 2019-11-25 by jwhui