openthread/openthread #4100: Max allowed children

seungigi, 2019-08-16:
Hi all,

I'm trying to build a large scale Thread network which has about 500 children per router. According to the OpenThread webpage (https://openthread.io/guides/thread-primer/node-roles-and-types), within a single network, each router can have up to 511 children.

But the OpenThread implementation seems not to support that many children. When I checked the source code (include/openthread/thread_ftd.h), there are some related functions such as;

uint8_t otThreadGetMaxAllowedChildren(otInstance *aInstance);
otError otThreadSetMaxAllowedChildren(otInstance *aInstance, uint8_t aMaxChildren);

The issue is that these functions use uint8_t. I think to support more than 255 children it should use uint16_t.

Regards,
Seung

bukepo, 2019-08-19:
Yes, we should change that to `uint16_t`. Besides,  there are two threads in openthread-users group:

https://groups.google.com/forum/#!topic/openthread-users/wpUPlhCEYlI
https://groups.google.com/forum/#!topic/openthread-users/QEFC2kYLGMM

They might be helpful as well.

jwhui, 2019-08-20:
Duplicate of #3979.

PR #4118 (2019-08-23): [child-table] add support for more than 64 children
Fixes #3979 
Fixes #4100
closed 2019-08-20 by jwhui