openthread/openthread #4105: Add feature to determine if multicast packets should be looped back to the local sockets.

huamenggg, 2019-08-19:
**Is your feature request related to a problem? Please describe.**
I'm implementing website topology graph of OpenThread on OpenWRT, and I need to get the information of the whole network. But I found that multicast packets won't be looped back to the local sockets.

**Describe the solution you'd like**
We can set or read a boolean integer argument that determines whether sent multicast packets should be looped back to the local sockets. like the IP_MULTICAST_LOOP (since Linux 1.2)

bukepo, 2019-08-19:
From https://www.tldp.org/HOWTO/Multicast-HOWTO-6.html, it seems this behavior is per socket based. I think we should extend the `UdpSocket` to include some flags.

bukepo, 2019-08-19:
Another issue needs to consider is multicast datagrams can be processed by multiple sockets of a single host. #3398 doesn't take into multicast into account. After we introduced UDP flags, we may consider enhance the multicast situation as well.

bukepo, 2019-08-20:
Thoughts? @jwhui

bukepo, 2019-08-20:
@librasungirl

jwhui, 2023-06-27:
Closing stale issue.

closed 2023-06-27 by jwhui