openthread/openthread #4128: Improve multicast, allow children to listen to src:0xffff or maybe redefine Link-Local to Logical-Local

musalk-dk, 2019-08-27:
2 issues closely related.
1. A Child does not listen to 0xffff broadcast for ff02::1 or ff03::1 only if it the source is it's router. meaning even if it was in range and received the multicast it will ignore it leaving it to only react to 1 of many possible received radio transmissions containing a valid message for the child.

2. A message to ff02::1 or ::2 is not actually as described "Link-Local — all interfaces reachable by a single radio transmission" for children - as a child within range will not receive a message sent from a child with different parent even tho they are i radio range. Making the term for "ff02::?" not Link-Local but Logical-Local.

A solution for 1.: 
Allow children to receive messages from src: 0xffff this will drastically improve delivery assurance of multicasts since the child will most likely hear the message from others before it's own router sends it there by also decreasing delivery latency.
A solution for 2.:
Is closely related to point 1. Sending ff02::? local link multicasts to dst:0xffff and listening to it will allow children to receive messages from it's non parental related neighbors.

Alternatives i have looked at is the raw module for making a actual local link radio transmission meaning it would be received by all neighboring devices regardless of heritage however there is no documentation on it's usage. Like can one use the raw api concurrent with thread?

**Additional context**
So far very happy with open thread and the api structure falling right with in my coding style but my implementation plan of an speedy efficient distribution of data blobs was kind of destroyed with "Link-local" actually being "Logical-Local" :)

musalk-dk, 2019-08-29:
@jwhui thanks for answering so fast.

1. We are running all of our devices as FTD and i am pretty sure we are running RX always on. But i have to check if that was the specifically defined with the last test. 
But if i understand you correctly it dose not matter if the device is FTD and rx on always if the broadcast message comes from child with different parent. As according to you they will only listen to whom they have a "MLE" relationship with.
**Q: Is that correct?**

2. We require as low latency and low packet loss as possible. So we designed a distribution model based on ff02::1 local-link using different ways to ensure every device got the message without a complete/"controlled" flod as with ff03::1. However the design broke down with the addition of children due to devices not receiving otherwise in range and "valid" broadcast packets...

3. So i would state that the definition of scope:
"Link-Local — all interfaces reachable by a single radio transmission"
in the documentation/thread primer should be changed to Logical-local or have an addendum stating something like "all **connected**/**established** interfaces reachable by a single radio transmission"
Just a suggestion :)

4. Also in the primer on [https://openthread.io/guides/thread-primer/router-selection](url)
there is a the following text: 
"An End Device may also wish to establish a receive link with neighboring non-Parent Routers to improve multicast reliability. We'll learn more about this when we get to Multicast Routing."
However that is the last you see of that...
**Q: Perhaps you can spread some light on this?**

5. The data we need to disseminate does not need encryption only source verification so we were considering to send the packets as raw but cryptographically signed packets or time share the radio with a custom protocol. As we really like many of the features offered by thread.
**Q: A. Is it possible to use the raw link api concurrent with thread to support this or do we need to use the time share approach?**

jwhui, 2019-08-30:
@musalk-dk 

1. Because Thread requires IEEE 802.15.4 MAC security for all data traffic, Thread devices will generally only receive data traffic for neighbors where an MLE link is established.

2. One exception to the above is MLE packets, which are sent without IEEE 8021.5.4 MAC security enabled. MLE is used to establish a secure link, which is why MLE does not require MAC security. In the case of MLE, link-local scope is as defined in the Thread specification.

3. FTDs operating as children continuously listen for MLE Advertisement messages from any neighboring routers. Upon discovering a new neighboring router, the FTD child may use MLE Link Request/Accept to establish an MLE link for the purposes of receiving multicast messages.

4. If your application only requires link-local communication and does not require MAC security, then you could try sending "unsecure" traffic. I have pushed a [minor patch](https://github.com/openthread/openthread/compare/master...jwhui:udp-test) to the UDP CLI that should allow you to play with this feature. Note that this behavior is not a part of the Thread specification.

Hope that helps.