openthread/openthread #4159: [api] unify enable/disable API style

bukepo, 2019-09-11:
**Is your feature request related to a problem? Please describe.**
Currently we have two styles for enable/disable. Most uses two separate APIs. But `otLinkRawSetEnable(bool)` and `otLinkRawSrcMatchEnable(bool)` accepts a parameter to disable the functionality, shall we unify the style by adding two disable APIs?

jwhui, 2019-09-11:
The following APIs seems to follow the `SetEnabled(boo)` model:
- otChannelManagerSetAutoChannelSelectionEnabled
- otChannelMonitorSetEnabled
- otIp6SetEnabled
- otIp6SetMulticastPromiscuousEnabled
- otIp6SetReceiveFilterEnabled
- otIp6SetSlaacEnabled
- otLinkSetEnabled
- otLinkRawSetEnable
- otLinkRawSetPromiscuous
- otThreadSetEnabled
- otThreadSetRouterRoleEnabled
- otPlatRadioEnableSrcMatch
- otPlatRadioSetPromiscuous

bukepo, 2019-09-11:
Thanks! Which style is recommended for new APIs?

jwhui, 2019-09-11:
I think the preference has been the `SetEnabled(bool)` style, rather than two separate APIs.

bukepo, 2019-09-12:
Thank you for the confirmation!

closed 2019-09-12 by bukepo