openthread/openthread #4161: Do we allow attached node operate in promiscuous mode?

bukepo, 2019-09-12:
**Is your feature request related to a problem? Please describe.**
From https://github.com/openthread/openthread/blob/master/src/core/api/link_api.cpp#L300, we don't allow.
From https://github.com/openthread/openthread/blob/master/src/ncp/ncp_base.cpp#L1356, no check for this at all.

**Describe the solution you'd like**
I think it makes sense to support this, so that an attached node can still act as sniffer.

**Describe alternatives you've considered**
None

**Additional context**
None

bukepo, 2019-09-12:
@smides-nest

jwhui, 2019-09-12:
FWIW, many IEEE 802.15.4 radios do not support both promiscuous mode and auto generation of immediate acks.

closed 2020-07-17 by jwhui

simonlingoogle, 2020-07-17:
> FWIW, many IEEE 802.15.4 radios do not support both promiscuous mode and auto generation of immediate acks.

We actually had a plan to implement this in the future to support functional Thread stack and sniffing at the same time.

jwhui, 2020-07-17:
> > FWIW, many IEEE 802.15.4 radios do not support both promiscuous mode and auto generation of immediate acks.
> 
> We actually had a plan to implement this in the future to support functional Thread stack and sniffing at the same time.

@simonlingoogle , what was the plan to support radios that do not support promiscuous mode and auto-generation of immediate acks at the same time?

simonlingoogle, 2020-07-19:
> @simonlingoogle , what was the plan to support radios that do not support promiscuous mode and auto-generation of immediate acks at the same time?

It was actually less of a plan, but more of a thought or idea. 
The intention was to have similar user experience to other networks (e.g. Ethernet) in which developers can use `wireshark` to capture all packets while the network stack is still working. We expected it to be helpful for OT developers to diagnose issues on OTBR.

jwhui, 2020-07-20:
> It was actually less of a plan, but more of a thought or idea.
> The intention was to have similar user experience to other networks (e.g. Ethernet) in which developers can use `wireshark` to capture all packets in promiscuous mode while the network stack is still working. We expected it to be helpful for OT developers to diagnose issues on OTBR.

@simonlingoogle , I agree it can be useful. However, as mentioned previously, many existing radios do not support promiscuous mode and auto-generation of immediate acks at the same time. It may be that newer radios do support this, however.

PR #5245 (2020-07-17): [ncp] only allow promiscuous mode in link raw mode
Many existing IEEE 802.15.4 radios do not support both promiscuous
mode and auto generation of immediate ACKs. This commit prevents
promiscuous mode when the interface is up.

Resolves #4161