openthread/openthread #4168: Issue with "Build a Thread network with nRF52840 boards and OpenThread" installation

embeddedRSA, 2019-09-13:
When running the command 
`$ sudo /usr/local/sbin/wpantund -o Config:NCP:SocketPath /dev/ttyACM0  \
        -o Config:TUN:InterfaceName utun7 \
        -o Daemon:SyslogMask " -info"`
i get this error:
`
wpantund[6831]: Starting wpantund 0.08.00d (Sep 12 2019 14:41:58) . . .

wpantund[6831]: 	SOURCE_VERSION = 0.07.01-319-g1ec2b73

wpantund[6831]: 	BUILD_VERSION = 0.07.01-319-g1ec2b73

wpantund[6831]: Configuration file "/usr/local/etc/wpantund.conf" read.

wpantund[6831]: Ready. Using DBUS bus ":1.35"

wpantund[6831]: Running as root without dropping privileges!

wpantund[6831]: SpinelNCPInstance-Protothreads.cpp:412: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)

wpantund[6831]: Resetting and trying again... (retry 1)

wpantund[6831]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)

wpantund[6831]: Resetting and trying again... (retry 2)

wpantund[6831]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)

wpantund[6831]: Resetting and trying again... (retry 3)

wpantund[6831]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)

wpantund[6831]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault 
state.

wpantund[6831]: State change: "uninitialized" -> "uninitialized:fault"

^C

Caught SIGINT!

wpantund[6831]: select() errno="Interrupted system call" (4)

wpantund[6831]: Cleaning up. (gRet = 130)

wpantund[6831]: Stopped.

`

I have been following the ["Build a Thread network with nRF52840 boards and OpenThread"](https://codelabs.developers.google.com/codelabs/openthread-hardware/#0) guide. When installing on lubuntu (16.04.3 LTE), in (virtual-box, without the Oracle VM VirtualBox Extension Pack" i didn't encounter any issues. I installed the latest version of everything (Jflash, nRF command line tools, and ARM GNU toolchain) i didn't encounter any apparent errors.
Afterwards i followed step 3, and the repository scripts being adressed. Here i tried adding libredline-dev (i earlier tried without). this is the "wpantund/configure" output
[image]

Jflash seems to work, when flashing the example to the device. 

some inconsistencies i found with my installation (compared to the guide) is

1. no configure file `etc/wpantund.conf` can be found
2.  nrjfprog path after intallation is: /opt/nrfjprog (package installer placed it there)
3. as i understood in step 2, im supposed to place the extracted ARM GNU Toolchain archive in "opt/gnu-mcu-eclipse/arm-none-eabi-gcc/" i created the directories myself

jwhui, 2019-09-18:
@embeddedRSA, the wpantund symptoms you are experiencing typically indicates that wpantund is having trouble communicating with the NCP.

Have you validated that serial communication from Linux host to nRF52840 works?

embeddedRSA, 2019-09-20:
Hey jwhui,(sorry for late reply), no i haven't, but the device turned up as /dev/ttyACM*.. I suspect that the problem was because of lack of the VirtualBox expansion (which i'm not allowed to use).. either way. Im running linux ubuntu natively now and the issue is now non-existent.

closed 2019-09-20 by embeddedRSA