openthread/openthread #4173: Adding support for automatic ifconfig replacing command line

binartech, 2019-09-16:
I'm trying to automate the connection to the thread network using ot API.

I added the otIP6SetEnabled..... to the code.
form some reason I cant find, the linker doesnt find the module containing the function.

/mnt/c/openthread/examples/../examples/platforms/efr32mg21/system.c:149: undefined reference to `otIp6SetEnabled'
collect2: error: ld returned 1 exit status

The example is a modification of the cli based example, however the cli works and the function is used by the cli, but the linker doesnt find it when used in the system.c file .

Any advice on this ?

Platform: EFR32MG21

jwhui, 2019-09-17:
@binartech, `otIp6SetEnabled()` only controls the IPv6 interface. You also need to call `otThreadSetEnabled()` to enable the Thread-related protocols.

Take a look at our [Deveoping with OpenThread APIs](https://codelabs.developers.google.com/codelabs/openthread-apis) and, in particular, [8. API: Configure the Thread network](https://codelabs.developers.google.com/codelabs/openthread-apis/index.html?index=..%2F..index#7).

Hope that helps.

binartech, 2019-09-17:
@jwhui yes, several calls must be added, the thing is that all the thread api is implemented in C++, and the example source is C. I wonder if theres flag in the make file that allows to work

jwhui, 2019-09-17:
@binartech, the OpenThread APIs defined in [include/openthread](https://github.com/openthread/openthread/tree/master/include/openthread) are C (not C++) APIs.

binartech, 2019-09-17:
@jwhui you are correct the header is C, but source code of the api is C++: 
[https://github.com/openthread/openthread/blob/master/src/core/api/ip6_api.cpp](url)
uses using namespace ot;
And looking at the cli.cpp its laso uses : namespace ot { 
So there's no way to use the api from 'C' unless there's a flag in the makefiles which will allow the build with plain 'C' ? 
Thanks for  helping,
by the way closed the issue accidentally

jwhui, 2019-09-17:
@binartech, if you are using the public OpenThread API, then everything can be done in C. You should not need to reference C++ APIs directly. Can you provide a concrete example of what you are trying to call?

binartech, 2019-09-17:
@jwhui sure, added the call to:
otIp6SetEnabled(sInstance, true);
The problem is in link time, no error when the file is compiled but at link time you get:
../../../examples/platforms/efr32mg21/libopenthread-efr32mg21.a(libopenthread_efr32mg21_a-system.o): In function `otSysInit':
/mnt/c/openthread/examples/../examples/platforms/efr32mg21/system.c:149: undefined reference to `otIp6SetEnabled'
collect2: error: ld returned 1 exit status
I suspect that when the libopenthread-efr32mg21.a was built in the early stage of the makefile theres a flag missing to be able to call the C OT functions. 
the code also includes the ip6.h which declarations are correct and at compile time theres no error but the problem is when the makefile mades the linking of the the compiled files.

jwhui, 2019-09-17:
A couple notes:

1. The link error is probably occurring when trying to link `ot-rcp`. By design, the RCP does not include OpenThread and, as a result, does not include `otIp6SetEnabled()`. Try removing [`--enable-radio-only`](https://github.com/openthread/openthread/blob/c0ef32766447ccfd0cf5e89650a574cfc6187f97/examples/Makefile-efr32mg21#L49) from `examples/Makefile-efr32mg21`.

2. The `otSys*` functions are shared across all of the build configurations, including FTD, MTD, and RCP. You may consider using `otIp6SetEnabled()` in the application code directly, as is done in [examples/platforms/efr32mg21/sleepy-demo/sleepy-demo-ftd/main.c](https://github.com/openthread/openthread/blob/c0ef32766447ccfd0cf5e89650a574cfc6187f97/examples/platforms/efr32mg21/sleepy-demo/sleepy-demo-ftd/main.c#L104).

Hope that helps.

binartech, 2019-09-17:
@jwhui Thanks for the great help, after removing the --enable-radio-only it worked.
having other issues but in other comment.

closed 2019-09-17 by jwhui