openthread/openthread #4185: openthread-core-config.h is included by mbedtls-config.h

e-rk, 2019-09-18:
Is there any reason why `openthread-core-config.h` is included in `mbedtls-config.h`?
I think that this include is unnecessary and causes us to copy all header files from `src/core/config` in order to use the libraries and mbedtls headers in an external project.

Should I make a PR to remove it?

jwhui, 2019-09-18:
Some of the defines in [third_party/mbedtls/mbedtls-config.h](https://github.com/openthread/openthread/blob/master/third_party/mbedtls/mbedtls-config.h) are conditional on [OpenThread build configurations](https://github.com/openthread/openthread/blob/bb56fa906a09408acc3787ae54969270e43a2454/third_party/mbedtls/mbedtls-config.h#L79). How would we ensure the appropriate build configs are defined without including `openthread-core-config.h`?

e-rk, 2019-09-19:
I thought at first that most of the definitions were enabled only by `common-switches.mk` and the `src/core/config` files provided only default values, most of which were set to `0`, but then I remembered about the `openthread-config-generic.h` generated by the `configure`.
Sorry for misuderstanding.

jwhui, 2019-09-20:
The values in [`src/core/config`](https://github.com/openthread/openthread/tree/master/src/core/config) specify default configuration values. I believe all of the `OPENTHREAD_CONFIG_*` values used by [`third_party/mbedtls/mbedtls-config.h`](https://github.com/openthread/openthread/blob/master/third_party/mbedtls/mbedtls-config.h) currently default to zero, so we could replace:
```c++
#include "openthread-core-config.h"
```
with:
```c++
#ifdef OPENTHREAD_PROJECT_CORE_CONFIG_FILE
#include OPENTHREAD_PROJECT_CORE_CONFIG_FILE
#endif
```

However, if we ever change any of the defaults to something non-zero, then that would not get pulled in. Thoughts?

e-rk, 2019-10-29:
We are now using a custom `mbedtls-config.h` that does not rely on `OPENTHREAD_CONFIG` definitions.
Closing.

closed 2019-10-29 by e-rk