openthread/openthread #4187: Not joining Join failed [Security]

binartech, 2019-09-18:
Any way to diagnose why the join fails ?
 networkname pmcaiotnet
Done
> 
> extpanid 1122334455667788
Done
> 
> panid 0x1122
Done
> 
> masterkey 656D62657220454D3235302063686970
Done
> 
> ifconfig up
Done
> 
> thread start
Done
commissioner start
Commissioner: petitioning
Done
> Commissioner: active
> commissioner joiner add * PMCA3000
Done

on the joiner side:
ifconfig up
Done
> 
> joiner start PMCA3000
Done
> 
> Join failed [Security]

binartech, 2019-09-18:
The scan on the joiner side returns

| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
> 
> | 1 | pmcaiotnet       | 1122334455667788 | 1122 | a6a0dce23e8975d8 | 11 | -37 | 255 |

binartech, 2019-09-19:
Another comment, I'm using firmware built with flags commissioner=1 , joiner=1. Loaded same firmware in both joiner and commissioner. Is this a correct implementation or joiner should have joiner=1 flag but not commissioner=1? ( and viceversa)

binartech, 2019-09-23:
Solved. Added code to join from code instead from CLI. Seems that the instance initialized in the code using:  sInstance=otInstanceInitSingle(); interferes with the instance that the CLI initialized and causing the join failed.

jwhui, 2019-09-23:
@binartech, thanks for the update. Are you using code straight from GitHub or do you have custom code? AFAIK, OpenThread's existing CLI component does not initialize its own OpenThread instance.

binartech, 2019-09-23:
@jwhui to the examples of the Github code I started to add some custome code. ( Im using the EFR32MG21 examples)
at the app/cli/main.c theres a line:  otCliUartInit(instance);
so Cli will use this instance. 
Without looking before this I added sInstance=otInstanceInitSingle(); to the system.c and that for sure interfere with the above initialization. 
Removing the line I added to the system.c it solved the problem. I think if I want to interact without problems I have to modify the app/cli/main.c so I can use the instance initialized in this file and continue adding API functionality while preserving the ability to use the CLI.

jwhui, 2019-09-23:
@binartech, [`examples/apps/cli/main.c`](https://github.com/openthread/openthread/blob/a1cffd2545efbdc38ee8c59b490567f5c3f82608/examples/apps/cli/main.c#L101) already includes a call to `otInstanceInitSingle()`. Adding another call to `otInstanceInitSingle()` somewhere else will certainly cause issues.

closed 2019-09-23 by jwhui