openthread/openthread #4197: Not able to ping though borderrouter to link-local

embeddedRSA, 2019-09-20:
Hi, i have set up my border router and i amm able to ping to a global server by using the well-known prefix:
```
> ping 64:ff9b::0808:0808
> 16 bytes from 64:ff9b:0:0:0:0:808:808: icmp_seq=55 hlim=51 time=26ms
```
My problem is pinging link-local addresses such as this one:

```
wlo1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.3.144  netmask 255.255.255.0  broadcast 192.168.3.255
        **inet6 fe80::593d:100:f1d8:44f**  prefixlen 64  scopeid 0x20<link>
        ether ac:b5:7d:26:2f:ba  txqueuelen 1000  (Ethernet)
        RX packets 180966  bytes 161395680 (161.3 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 117698  bytes 18161299 (18.1 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
(this is on my border router laptop)
when i try to ping from a node a respond never returns

```
>state
leader
Done
> ping fe80::8817:3ff:fe80:61fa
```
Is what im trying to do possible?

jwhui, 2019-09-20:
It seems that you are trying to ping a link-local address of a non-Thread interface on the border router. IPv6 link-local addressing explicitly prohibits this use case.

[RFC 4007](https://tools.ietf.org/html/rfc4007) states:
```
   o  Link-local scope, for uniquely identifying interfaces within
      (i.e., attached to) a single link only.
```
[RFC 4291](https://tools.ietf.org/html/rfc4291) states:
```
   Routers must not forward any packets with Link-Local source or
   destination addresses to other links.
```

embeddedRSA, 2019-09-23:
in that case, how are you supposed to ping non mesh adresses on the local network, and ping a specific node from the outside.

jwhui, 2019-09-23:
You need to use an IPv6 address with a larger (e.g. global) scope.

On the OpenThread Border Router, you can configure prefixes that have global scope for the interfaces you want to communicate between.

For example:
```bash
$ sudo ip -6 addr add fd11:22:33:1::1/64 dev wpan0
$ sudo ip -6 addr add fd11:22:33:2::1/64 dev wlo1
```

embeddedRSA, 2019-09-24:
would that be the interfaces between two border routers in my case?  such as "veth159d65" which appears when running docker (method used in an example at the openthread site), and likewise the interface used on my raspberry pi.