openthread/openthread #4199: No serial output or responses from EFR32MG12 BRD4161A when executing example apps

vijayrao1988, 2019-09-20:
I am using a Mighty Gecko 19dBm (BOARD4161A) with the WSTK. The code builds fine and I can load the output hex file into the flash using JLinkExe. However, there is no output on the serial interface.

The posix build executes perfectly fine.

Looking at the user-guide for BRD4161A, Page 20, Section 5.6 on Virtual COM Port, I think there might be a mssing VCOM_ENABLE. It might be needed to enable PA5 (GPIO) - https://www.silabs.com/documents/public/user-guides/ug260-brd4161a.pdf

Any suggestions on what could be wrong and how to get this sorted?

vijayrao1988, 2019-09-23:
I noticed that the hal-config.h file doesn't contain some important lines:
(1) lines regarding HAL_BUTTON_ENABLE and HAL_BUTTON_COUNT
(2) lines regarding HAL_VCOM_ENABLE

I updated the hal-config.h file from an example Simplicity Studio project which seemed to have these lines.

I now get the following error when I issue the make command:
~/openthread/examples/../third_party/silabs/gecko_sdk_suite/v2.6/hardware/kit/common/bsp/bsp_init.c:36:28: fatal error: retargetserial.h: No such file or directory
 #include "retargetserial.h"

binartech, 2019-09-23:
@vijayrao1988 Im using the EFR32MG21 kit which I think is very similar to the one you have.
at build have you used the following line:
make -f examples/Makefile-efr32mg12 BOARD=BRD4161A
before build or rebuild make sure to start from clean project

make -f examples/Makefile-efr32mg12 BOARD=BRD4161A clean

[https://github.com/openthread/openthread/blob/master/examples/platforms/efr32mg21/README.md](url)
The Uart used is the usart1 which is the serial 1 in the console of simplicity studio

vijayrao1988, 2019-09-23:
@binartech I did do a make clean before the make -f ....efr32mg12 BOARD=BRD4161A command. I can clearly see VCOM and BUTTON missing in the hal-config file for BRD4161A. These 2 are mentioned for efr32mg21 boards.

When I add these (VCOM and BUTTON), I get build errors related to HAL drivers/libraries.

binartech, 2019-09-23:
@vijayrao1988 
Not looked at the buttons, just at the VCOM

In your hal-config.h for your BRD4161a look for the 
// $[VCOM]
// [VCOM]$
 
and replace it with 

// $[VCOM]
#define HAL_VCOM_ENABLE                       (1)

#define BSP_VCOM_ENABLE_PIN                   (5U)
#define BSP_VCOM_ENABLE_PORT                (gpioPortA)

// [VCOM]$

VCOM should be enabled and you should be able to access CLI

vijayrao1988, 2019-09-24:
An earlier issue on the silabs forums similar to the retargetserial.h missing file error I am getting can be seen here - https://www.silabs.com/community/software/simplicity-studio/forum.topic.html/simplicity_studiopr-kRrX

Is there any connection between these two issues?

nigelg-silabs, 2019-10-04:
We are using VCOM on 4161 without an issue.  Can you confirm what terminal emulator are you using - Simplicity Studio? If you are it should be visible in the Serial 1 window.

You should set PA5 as you suggest to enable VCOM 

https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2017/12/03/setting_up_serialou-fJAC

It may be useful to turn on additional logging in OpenThread which outputs via Segger RTT.  The freely downloadable Segger J-Link RTT Viewer is available on their website in the J-link Software and Documentation Pack. Add FULL_LOGS=1 as an argument to the make command to enable logging via RTT.   You can selectively disable and enable module logging in the examples/common-switches.mk file.
In Segger's RTT Viewer you may need to configure a Search Range to find the Control Block with
"0x20000000 0x00040000" when you connect.

jwhui, 2019-11-04:
Closing stale issue.

closed 2019-11-04 by jwhui