openthread/openthread #4200: [mac] assert seen while SED performed data poll

LuDuda, 2019-09-22:
In our stress tests where nodes constantly send some packets quickly (mostly ICMPv6 multicast ones) several SEDs end up with the same assert [here](https://github.com/openthread/openthread/blob/6fd3a24a43de602ac7b22a391a761beb107cf870/src/core/mac/mac.cpp#L1266). I believe this assert has been introduced with #3915. The SED devices have a polling interval of around 5s and they try to send frames with 200ms interval.

Any idea, how this could happen? Because it seems that the platform notifies about TX frame being sent that was not the Data Poll but rather the ICMPv6 ping with the 0xFFFF destination address (however take into account that the firmware was built with Os optimization).

GDB logs:
```
(gdb) target remote :54517
Remote debugging using :54517
0x00007e30 in ot::Mac::Mac::HandleTransmitDone (this=0x2000a240 <ot::gInstanceRaw+37976>, aFrame=..., aAckFrame=0x0 <__isr_vector>, aError=<optimized out>) at /build/openthread/examples/../src/core/mac/mac.cpp:1266
1266	/build/openthread/examples/../src/core/mac/mac.cpp: No such file or directory.
(gdb) where
#0  0x00007e30 in ot::Mac::Mac::HandleTransmitDone (this=0x2000a240 <ot::gInstanceRaw+37976>, aFrame=..., aAckFrame=0x0 <__isr_vector>, aError=<optimized out>) at /build/openthread/examples/../src/core/mac/mac.cpp:1266
#1  0x00007e7c in ot::Mac::Mac::HandleTransmitDone (this=0x2000a240 <ot::gInstanceRaw+37976>, aFrame=..., aAckFrame=0x0 <__isr_vector>, aError=<optimized out>) at /build/openthread/examples/../src/core/mac/mac.cpp:1307
#2  0x00017f9a in nrf5RadioProcess (aInstance=aInstance@entry=0x20000de8 <ot::gInstanceRaw>) at /build/openthread/examples/../examples/platforms/nrf52840/radio.c:626
#3  0x0001834c in otSysProcessDrivers (aInstance=aInstance@entry=0x20000de8 <ot::gInstanceRaw>) at /build/openthread/examples/../examples/platforms/nrf52840/system.c:137
#4  0x000002f8 in main (argc=0, argv=0x0 <__isr_vector>) at /build/openthread/examples/../examples/apps/cli/main.c:110
(gdb) info locals
dstAddr = {
mShared = {
mShortAddress = 65535,
mExtAddress = {
<otExtAddress> = {
m8 = "\377\377\000 \000\000\000"
}, <No data fields>}
},
mType = ot::Mac::Address::kTypeShort
}
(gdb) info args
this = 0x2000a240 <ot::gInstanceRaw+37976>
aFrame = @0x2000b300: {
<ot::Mac::Frame> = {
<otRadioFrame> = {
mPsdu = 0x2000b319 <sTransmitPsdu+1> "Iر4\022\377\377}\204j\322hc\t\372\rKB\001",
mLength = 64,
mChannel = 11 '\v',
mInfo = {
mTxInfo = {
mAesKey = 0x2000a1b8 <ot::gInstanceRaw+37840> "z\vz\205췥0\356\317\034ƽx\215\251",
mIeInfo = 0x0 <__isr_vector>,
mMaxCsmaBackoffs = 32 ' ',
mMaxFrameRetries = 3 '\003',
mIsARetx = false,
mCsmaCaEnabled = true
},
mRxInfo = {
mTimestamp = 536912312,
mRssi = 32 ' ',
mLqi = 3 '\003',
mAckedWithFramePending = false
}
}
}, <No data fields>}, <No data fields>}
aAckFrame = 0x0 <__isr_vector>
aError = <optimized out>

```

abtink, 2019-09-23:
@LuDuda, I will take a look to see if I can find anything in core (mac implementation).

abtink, 2019-09-23:
@LuDuda , I did a bit of investigation on this. Could not see/find anything in Mac/SubMac code. The `Mac` operations are done in serial order, and we only start next operation after finishing the previous one. In case of all Tx operations, we pass the frame to radio platform and should wait for `TxDone` callback from radio platform layer before finishing operation. The gdb output indicates the PSDU is a data frame (not a data poll).

I tried a similar stress test (ping to multicast while sending polls) in simulation for longer time (and on another hw platform for a short time) but could not repro/see the issue. 

Is there a possibility that radio driver happens to pass a previous frame in tx done callback? The `assert` check itself was added as part of the https://github.com/openthread/openthread/pull/3915. Before that the code would have accepted any frame from `TxDone` path  (the data poll itself was a `message` in the send queue of `MeshForwarder`).

abtink, 2019-09-23:
@LuDuda , I checked the radio driver code (nrf52840) and all seems fine to me as well. 
Would it be possible to run the test with OT logging at `DEBG`? I think that would help give us good info about this.

LuDuda, 2019-09-24:
@abtink thank you a lot for taking time on this. I will try to enable logs and replicate the bug on a smaller set up on my desk to see if I can capture it again.

simonlingoogle, 2019-10-22:
I have actually met with the same `assert`. 
It happens when I after I run  `scan` command for multiple times in a heavy traffic. 
I think I can reproduce this issue and I will look into it when I am available.

simonlingoogle, 2019-10-22:
I added a log at the beginning and end of `Mac::BeginTransmit`:
```
# beginning
otLogCritMac("BeginTransmit Enter OP=%d, AR=%d", mOperation, sendFrame.GetAckRequest());
# end
otLogCritMac("BeginTransmit Exit OP=%d, AR=%d, E=%d", mOperation, sendFrame.GetAckRequest(), error);
```
Before assert failed, the log is like:
```
[CRIT]-MAC-----: BeginTransmit Enter OP=6, AR=0
[CRIT]-MAC-----: PrepareDataRequest: E=11
[CRIT]-MAC-----: BeginTransmit Exit OP=6, AR=0, E=11
[CRIT]-MAC-----: HandleTransmitDone: OP=6, E=11
```
Because `PrepareDataRequest` returned `OT_ERROR_ABORT`, `sendFrame` was not property prepared. I think `Mac::HandleTransmitDone` should handle the case when `error != OT_ERROR_NONE` more thoughtfully.

abtink, 2019-10-22:
@simonlingoogle nice debugging.

simonlingoogle, 2019-10-22:
@abtink Thanks. I can reproduce this issue within minutes using the simulator I recently developed: [go/otns-dd](http://go/otns-dd). Please take a look and give feedbacks if you have time.

abtink, 2019-10-22:
Just submitted which should address this: https://github.com/openthread/openthread/pull/4264

abtink, 2019-10-22:
Submitted PR https://github.com/openthread/openthread/pull/4265 replacing  https://github.com/openthread/openthread/pull/4264. This should help address this issue and other situations where the previous frame is being used in TxDone callbacks

abtink, 2019-10-24:
The https://github.com/openthread/openthread/pull/4265 is merged and should help address this.

Thanks all for help with this. :)

This was a tricky issue (which I think was a long-standing existing one). I think the new assert in data poll handling path helped expose it.

closed 2019-10-24 by abtink

LuDuda, 2019-10-24:
@abtink @simonlingoogle thank you for this effort!

simonlingoogle, 2019-10-25:
@LuDuda @abtink good news I can not reproduce this issue now