openthread/openthread #4204: verifiying if node already joined the network

binartech, 2019-09-23:
Is there any API that checks the status of the  join of node ?
The Idea is on startup to verify if the node was already commissioned and will connect to the network after reset of it was never commissioned and required to join the network ?
I saw the otJoinerGetState but this function if I understood correctly will check the status dring the join process and I' would like to know the status for example after cycling the node power in which case theres no need to join again.

jwhui, 2019-09-23:
@binartech, see [otDatasetIsCommissioned()](https://openthread.io/reference/group/api-thread-general#otdatasetiscommissioned).

binartech, 2019-09-23:
@jwhui 
great thanks, will look at it, thanks and sorry for so many questions. Trying to move from SL-Thread to openthread. In SL-thread everything works but you dont see the inner things and additional possibilities  that can be achieved using open thread.

jwhui, 2019-09-23:
@binartech, thanks for the additional context. Let us know if you have additional questions.

closed 2019-10-25 by jwhui