openthread/openthread #4208: Sleepy-demo in 915MHz band.

mselser, 2019-09-25:
Hello, I'm currently trying to make the Sleepy-demo with the EFR32MG12 (board 4170A) example use the 915MHz band, but I can't find the way to do it. The only change I made to the example (in both mtd and ftd) code is this:

`/* Set Channel to 4 */
    aDataset.mChannel                      = 4;
    aDataset.mComponents.mIsChannelPresent = true;`

Instead of this:

`/* Set Channel to 15 */
    aDataset.mChannel                      = 15;
    aDataset.mComponents.mIsChannelPresent = true;`

Because channel 4 corresponds to the band I'm trying to use and channel 15 corresponds to the 2.4GHz band.

After compiling and running, the channel command returns this:
`> channel
11
Done`
And the state command returns this:
`> state
detached
Done
`
What else should I do to make it work?

Thank you very much.

Matías.

zhanglongxia, 2019-09-25:
Hi Matías,

The channel range is from 1 to 10 in 915MHz band. The center frequency of these channels is: Fc = 906 + 2 (k – 1) in megahertz, for k = 1, 2, ..., 10.  You may need to set the channel in the channel range.

Thanks
Zhanglong

jwhui, 2019-09-25:
@zhanglongxia, it appears @mselser is setting the channel to 4, which is in the sub-GHz range.

zhanglongxia, 2019-09-25:
@jwhui I saw the current channel is set to 11. It seems the sub-GHz  is not enabled.   @mselser could you check if you enabled sub-GHz.

mselser, 2019-09-25:
That's probably the issue, I don't know where to enable the sub-GHz. What file do I have to edit?

Thanks

zhanglongxia, 2019-09-26:
Hi mselser:

I checked the code, the sub-GHz is enabled by default on board 4170A. So you don't need to enable it. 

Could you add the following code to Sleepy-demo to set the channel mask, then compile and run your code again to see if the channel could be changed to 4.
```
aDataset.mChannelMask=0x07fffffe ;
aDataset.mComponents.mIsChannelMaskPresent = true; 
```

If it still doesn't work, could you follow the following steps to check if the sub-GHz is working:
<1> Compile image `ot-cli-ftd ` for board 4170A using the following command:
```
make -f examples/Makefile-efr32mg12 BOARD=BRD4170A CHANNEL_MANAGER=1
```
<2> Download image`ot-cli-ftd` to your board 4170A.
<3> Run the following commands:
```
> channel 15
Done
> panid 1
Done
> ifconfig up
Done
> thread start
Done
> 
>  channel manager change 4
Done
```
<4> Wait about 2 minutes and check if the channel is changed to 4.
```
> channel
4
Done
```

Thanks
Zhanglong

mselser, 2019-09-26:
Hello,
I tried the first option, but it didn't work. 
After that, I followed your instructions with the client example and, after compiling and flashing, I got this error:
`> channel 15`
`Done`
`> panid 1`
`Done`
`> ifconfig up`
`Done`
`> thread start`
`Done`
`>    `
`> channel manager change 4`
`Error 6: Parse`

It seems that the channel manager is not working, but I made sure to compile with this command:

`make -f examples/Makefile-efr32mg12 BOARD=BRD4170A CHANNEL_MANAGER=1`

Thank you for your help and patience

Matias.

zhanglongxia, 2019-09-27:
Hi Matias,

It seems the channel manager is not enabled.  You may need to clean images first, then compile  the image again. You can follow the following commands to compile the image:
```
make -f examples/Makefile-efr32mg12 BOARD=BRD4170A clean
make -f examples/Makefile-efr32mg12 BOARD=BRD4170A CHANNEL_MANAGER=1  CHANNEL_MONITOR=1 
```

Thanks
Zhanglong

jwhui, 2019-09-27:
@zhanglongxia, any idea why the Operational Dataset method doesn't work?

zhanglongxia, 2019-09-28:
@jwhui I don’t have the dev board with me now. So I enabled both the sub-GHz and 2.4GHz on platform posix and added the function “[setNetworkConfiguration()](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32mg12/sleepy-demo/sleepy-demo-ftd/main.c#L138)” to [examples/apps/cli/main.c](https://github.com/openthread/openthread/blob/master/examples/apps/cli/main.c#L106). The platform posix can switch the channel to 4 when  the “`aDataset.mChannel`” is set to 4. It means the Operational Dataset is working properly. I don’t know why it doesn’t work on the platform efr32mg12.  I will debug this issue when I get the dev board in the next week.

mselser, 2019-09-30:
Hello

I compiled and flashed again but after starting thread, the board stops responding. I'm not sure what's causing this. On the other hand, I'd like to try this with the 4164A board. Is there support for this board?
Thanks.

zhanglongxia, 2019-09-30:
@mselser The PR [#3922](https://github.com/openthread/openthread/pull/3922) add an [assert]((https://github.com/openthread/openthread/blob/master/examples/platforms/efr32mg12/radio.c#L212)) to check if the first channel equals to the minimum channel for sub-GHz. The [minimum channel](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32mg12/radio.c#L271) is set to 1. However, the radio config file sets the [fist channel](https://github.com/openthread/openthread/blob/master/third_party/silabs/rail_config/rail_config.c#L162) to 0. This mismatch makes the function [assert](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32mg12/radio.c#L212) goes into a dead loop. I have submitted a PR [#4214](https://github.com/openthread/openthread/pull/4214) to resolve this issue.

After this issue is resolved, the dev board 4170A should be able to switch to channel 4.
Here are the detailed steps I did on my dev board 4170A:

Compile:
```
openthread $ ./bootstrap
openthread $ make -f examples/Makefile-efr32mg12 COMMISSIONER=1 JOINER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 BOARD=BRD4170A
```
Flash the image:
```
openthread $ arm-none-eabi-objcopy -O ihex ./output/efr32mg12/bin/sleepy-demo-mtd  ./output/efr32mg12/bin/ot-cli-ftd.hex
openthread $ JLinkExe -device EFR32MG12PxxxF1024 -speed 4000 -if SWD -autoconnect 1 -SelectEmuBySN 440085380
J-Link>loadfile ./output/efr32mg12/bin/ot-cli-ftd.hex
J-Link>r
J-Link>q
```
Open terminal to the device:
```
openthread $  screen /dev/tty.usbmodem0004400853801  115200
sleepy-demo-ftd started
sleepy-demo-ftd changed to leader

> state
leader
Done
> channel
4
Done
> 
```

mselser, 2019-10-01:
Hello. Thank you very much for your patience.

I followed step by step what you posted. My outcome was the following:
`sleepy-demo-ftd started`
`> state`
`detached`
`Done`

Which I think means that the network is not being created correctly.

So, I tried flashing the ot-cli-ftd example. When I did this, I found out that if the initial dataset configuration (with dataset init new) had a channel number corresponding to the 2.4GHz band, the network was created succesfully and I could get two boards to connect with each other. However, if the initial channel number was a channel corresponding to the 915MHz band, the connection didn't work. So I run a scan command in both cases, and I could verify that the network was successfully created in the first case, and in the second case the scan came out empty. Can you help me with this?

Again, thanks for your patience. 

Matías.

nigelg-silabs, 2019-10-04:
@mselser The sleepy app has a hard coded pan and channel to make demonstrations easy. By default it will go to channel 15 on the 2.4Ghz channel. This is done at initialisation so you may need to look at this.

mselser, 2019-10-04:
@nigelg-silabs I know, I changed the default channel to 4 (in both mtd and ftd). And after I checked that this didn't work, I did the client example thing I mentioned above.

zhanglongxia, 2019-10-07:
@mselser Sorry for the late reply. I double checked that if we apply the PR #4214 to the OpenThread code and follow my experiment steps above, the dev board 4170A should be able to switch to channel 4. The terminal of your device shows that the state is **detached**. I think it is caused by the previous stored info. You may need to run the command "**factoryreset**" to erase the stored the info.

closed 2019-10-16 by mselser