openthread/openthread #4237: Warnings appear with newest commit

AuSchell, 2019-10-11:
Hello everyone

I have recently updated my fork of openthread from commit eb5b42a86e8bc58e28b6290045457ad2c4caa223 to 862a2b3d02108cd07f604b25e199c9280936168a, and since then numerous warnings are occuring when building the code and i would like to get rid of them.

Here they are:

```
/third_party/nlbuild-autotools/repo/automake/post/rules/coverage.am:71: Warnung: Die Befehle für das Ziel „coverage-recursive“ werden überschrieben
/third_party/nlbuild-autotools/repo/automake/post/rules/coverage.am:71: Warnung: Alte Befehle für das Ziel „coverage-recursive“ werden ignoriert
/third_party/nlbuild-autotools/repo/automake/post/rules/pretty.am:69: Warnung: Die Befehle für das Ziel „pretty-recursive“ werden überschrieben
/third_party/nlbuild-autotools/repo/automake/post/rules/pretty.am:69: Warnung: Alte Befehle für das Ziel „pretty-recursive“ werden ignoriert
/third_party/nlbuild-autotools/repo/automake/post/rules/pretty.am:69: Warnung: Die Befehle für das Ziel „pretty-check-recursive“ werden überschrieben
/third_party/nlbuild-autotools/repo/automake/post/rules/pretty.am:69: Warnung: Alte Befehle für das Ziel „pretty-check-recursive“ werden ignoriert
/third_party/nlbuild-autotools/repo/automake/post/rules/pretty.am:73: Warnung: Die Befehle für das Ziel „pretty“ werden überschrieben
/third_party/nlbuild-autotools/repo/automake/post/rules/pretty.am:73: Warnung: Alte Befehle für das Ziel „pretty“ werden ignoriert
/third_party/nlbuild-autotools/repo/automake/post/rules/pretty.am:78: Warnung: Die Befehle für das Ziel „pretty-check“ werden überschrieben
/third_party/nlbuild-autotools/repo/automake/post/rules/pretty.am:78: Warnung: Alte Befehle für das Ziel „pretty-check“ werden ignoriert
/third_party/nlbuild-autotools/repo/automake/post/rules/headers.am:45: Warnung: Die Befehle für das Ziel „install-headers-recursive“ werden überschrieben
/third_party/nlbuild-autotools/repo/automake/post/rules/headers.am:45: Warnung: Alte Befehle für das Ziel „install-headers-recursive“ werden ignoriert
/third_party/nlbuild-autotools/repo/make/post/rules/help.mak:33: Warnung: Die Befehle für das Ziel „help-local“ werden überschrieben
/third_party/nlbuild-autotools/repo/make/post/rules/help.mak:33: Warnung: Alte Befehle für das Ziel „help-local“ werden ignoriert
/third_party/nlbuild-autotools/repo/make/post/rules/repos.mak:218: Warnung: Die Befehle für das Ziel „help-repos-local“ werden überschrieben
/third_party/nlbuild-autotools/repo/make/post/rules/repos.mak:218: Warnung: Alte Befehle für das Ziel „help-repos-local“ werden ignoriert
```

I use a NRF52840-DK board and build with `$ make -f examples/Makefile-nrf52840`. I am using Eclipse IDE for C/C++ Developers version 4.10.0. I have used the /script/bootstrapp to update all tools, so they should be at the newest version. My OS is Ubuntu 18.04.2 LTS.

Does someone know why those warnings appear? My code is still build, so it is not rather an annoyance than an obsticle, but i want to understand why this is happening.

Thank you for your answers.

jwhui, 2019-10-15:
@AuSchell, I do not see those compiler warnings. Have you tried on a fresh clone of the OpenThread repo?

AuSchell, 2019-10-15:
@jwhui, yes i have, It was the same result. Maybe it is a conflict between versions of the tools. I list them here:

- GNU automake v1.15.1
- GNU Make v4.1
- g++ v7.4.0
- libtool 2.4.6-2
- lsb-release v9.20170808ubuntu1
- gcc-arm-embedded v7-2018q2-1~bionic1
- clang-format v6.0.0

jwhui, 2020-02-28:
@AuSchell, are you still experiencing these warnings?

I have not been able to reproduce nor has anyone else confirmed seeing similar warnings. Also, between eb5b42a86e8bc58e28b6290045457ad2c4caa223 to 862a2b3d02108cd07f604b25e199c9280936168a, there were not any changes to `third_party/nlbuild-autotools`.

AuSchell, 2020-03-04:
@jwhui, yes, i do. I'm now on commit 9a4098fdb518f056b7dd07cce515848bb99c40eb. 
I'm starting to getting used to it. ^^

jwhui, 2021-04-07:
Closing stale issue. Using CMake/Ninja avoids these warnings.

closed 2021-04-07 by jwhui