openthread/openthread #4240: EID-to-RLOC cache errors seem to not be corrected.

hubertmis, 2019-10-13:
I've noticed a problem with frames not routed correctly to a SED. After some investigation I suspect that the messages are forwarded to an invalid router (not parent of the SED), because the cache errors are not detected or not corrected.

I'm attaching [the script](https://github.com/openthread/openthread/files/3721710/eid-to-rloc-cache-problem.zip) that more-less shows the real scenario I've used when I've found the problem. 

It looks that after a router times-out a SED and it receives a frame with a mesh header destination address matching the router's address, it shall detect EID-to-RLOC error and transmit an ICMP dest unreachable message. But no device in my test transmits such message. I suspect an issue in the implementation.

jwhui, 2019-10-15:
@hubertmis, thank you for raising this issue. I can help take a look at it soon.

jwhui, 2019-10-29:
@hubertmis, thank you again for raising this issue.

I have reviewed the script and the ping failure is expected. The OT CLI ping command only sends a single ICMP Echo Request packet. The ICMP Echo Request will go to the old / incorrect destination, the old parent will generate an ICMPv6 Error, and the EID-to-RLOC cache entry will be invalidated. As a result, dropping the initial packet sent to the old location is expected.

If I modify the test script to try sending the ICMPv6 Echo Request again after failure, the second ICMPv6 Echo Request will be properly routed and the ICMPv6 Echo Reply is received.

Thoughts?

librasungirl, 2019-10-29:
@hubertmis would you please check whether #4282 could help?
During the execution of the script
1) Leader would have one cache entry <SED-EID, RLOC16(DUT_ROUTER)> after it pings SED, when SED attaches to DUT_ROUTER;
2) Such <SED-EID, RLOC16(DUT_ROUTER)> still exists when SED attaches to Leader. When DUT_ROUTER pings SED, when the reply is forwarded by Leader, it would wrongly update its EidCache to <SED-EID, RLOC16(SED-L)> 
https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder_ftd.cpp#L595
3) When SED attaches to PARENT, and Leader sends ping to SED, Leader would think the dest is resolved however this packet would be dropped as SED is not a valid neighbor
https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder_ftd.cpp#L438
https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder_ftd.cpp#L449

In #4282, Router would remove the eid-rloc-cache entry which matches the addresses registered by child. Thus the 'wrongly update' in 2) won't happen.

jwhui, 2019-10-29:
@librasungirl, thanks for looking into this. I agree that #4282 will help the scenarios where the SED can invalidate the EID-to-RLOC cache entry in any new parent it attaches to. This is a useful optimization that we should incorporate.

That said, I do not think #4282 will address all scenarios - there are still scenarios where at least one message is required to generate an ICMPv6 Error to invalidate EID-to-CACHE entry.

jwhui, 2019-10-29:
> Another interesting phenomena noticed when testing with the modified script is:
> Given the topology L-R1-R2-SED, SED attaches to R2 but somehow L has one deprecated Eid-To-RLOC cache <SED, RLOC16(R1)>.
> Now L pings SED, this ping would be directed to R1 (without mesh header), and R1 would initiate Address Query, and finally the ping could be delivered to SED successfully.
> 
> Should we take it as some kind of error (generally R1 should send DstUnreachable?) or `optimization`? @jwhui

Thanks for reporting this observation! If the ICMPv6 Error is not getting generated, then I think there is a bug there. As for performing an Address Query when forwarding, I see this as a useful optimization because it avoids packet drops. I wonder if we should apply the same behavior to packets that were forwarded over multiple hops (i.e. with mesh header).

hubertmis, 2019-10-29:
@jwhui , @librasungirl Thanks for looking into this. I'll try to play with the script a little more to reproduce the real-life bug I've experienced with EID-to-RLOC cache errors. I'll update the script and verify if #4282 helps if I reproduce the issue. If I couldn't reproduce, I'll just close this ticket.

abtink, 2019-10-30:
Thanks @hubertmis, @librasungirl  for submitting and debugging this issue. This is indeed an interesting scenario.

We cover some of cache update scenarios from [test-021-address-cache-table.py](https://github.com/openthread/openthread/blob/master/tests/toranj/test-021-address-cache-table.py#L210). I added a new `toranj` test-case covering the situation from this issue (submitted https://github.com/openthread/openthread/pull/4293). Without the fix (current master), we see the message is looped around between two routers (since each have an address cache entry pointing to the other one for the child's address). With the fix from https://github.com/openthread/openthread/pull/4282 the test does pass ok.

jwhui, 2019-11-25:
@hubertmis, any further updates on this issue now that #4282 is merged?

jwhui, 2020-02-24:
Resolved by #4282

closed 2020-02-24 by jwhui