openthread/openthread #4245: Failed at unit test in the latest master branch

Irving-cl, 2019-10-15:
**Describe the bug**
When I run `make -f examples/Makefile-posix check` after bootstraping, I got the below errors:
```
error: ‘void* memset(void*, int, size_t)’ clearing an object of non-trivial type ‘class ot::Child’; use assignment or value-initialization instead [-Werror=class-memaccess]
     memset(&child, 0, sizeof(child));
                                    ^
In file included from /usr/local/google/home/irvingcl/Documents/git/openthread/examples/../src/core/thread/mle.hpp:48,
                 from /usr/local/google/home/irvingcl/Documents/git/openthread/examples/../src/core/common/settings.hpp:41,
                 from /usr/local/google/home/irvingcl/Documents/git/openthread/examples/../src/core/common/instance.hpp:61,
                 from /usr/local/google/home/irvingcl/Documents/git/openthread/examples/../tests/unit/test_platform.h:42,
                 from /usr/local/google/home/irvingcl/Documents/git/openthread/examples/../tests/unit/test_child.cpp:29:
/usr/local/google/home/irvingcl/Documents/git/openthread/examples/../src/core/thread/topology.hpp:379:7: note: ‘class ot::Child’ declared here
 class Child : public Neighbor, public IndirectSender::ChildInfo, public DataPollHandler::ChildInfo
       ^~~~~

```
The latest commit(ca798e2) could reproduce this problem
@abtink

abtink, 2019-10-15:
We have a `Child::Clear()` which should be used instead of `memset`. I will add a fix for this.

abtink, 2019-10-15:
Submitted "[test] update test_child to use Clear() instead of memset #4248"

jwhui, 2019-10-15:
Resolved by #4248.

closed 2019-10-15 by jwhui