openthread/openthread #4251: Cli example on EFR32MG12 BRD4164A

mselser, 2019-10-16:
Hello,

I tried to build the client example for the 4164A board, which is not in the supported list of boards.
In order to do so, I did de following:

1. Edit the makefile (Makefile-efr32mg12):
Added:
`ifeq ($(BOARD),BRD4164A)`
`MCU = EFR32MG12P433F1024GL125`
`else`

2. Created the corresponding folder for this board in the /examples/platforms/efr32mg12 directory. it was named "brd4164a".

3. Inside that folder, I created two files:

-  board_config.h: copied from the 4170A board. It only has defines that allow 915MHz and 2.4GHz communication.

- hal_config.h: this one was a bit trickier. This file depends on the MCU that the board has. So, to generate this file, I used Simplicity Studio. I opened a thread blank project and generated this file, and then coppied.

4. Finally, I verified if there was a folder in
/hardware/kit/EFR32MG12_$(BOARD)/config (this path is in makefile.am, inside the efr32mg12 folder) for BOARD=BRD4164A, which effectively was.

Afterwards, I compiled it with
` make -f examples/Makefile-efr32mg12 BOARD=BRD4164A`

And then, flashed it.

On the serial terminal, I had the following output:
`> dataset init new`
`Done`
`> dataset`
`Active Timestamp: 1`
`Channel: 1`
`Channel Mask: 07fffffe`
`Ext PAN ID: 18c39467b2b5e8e2`
`Mesh Local Prefix: fd96:1691:d260:d64c/64`
`Master Key: 5b0e2c8a413f4ac322d4802c7dfb8053`
`Network Name: OpenThread-3a8c`
`PAN ID: 0x3a8c`
`PSKc: cfc84abe9a803a750d8e8dd88fd34b9a`
`Security Policy: 0, onrcb`
`Done`
`> dataset commit active`
`Done`
`> ifconfig up`
And here the device stopped responding.

Also, I noticed that when i compiled the example with this line:
`make -f examples/Makefile-efr32mg12 BOARD=BRD4170A
The binaries are exactly the same!
`J-Link>loadfile ot-cli-ftd.hex`
`Downloading file [ot-cli-ftd.hex]...`
`Comparing flash   [100%] Done.`
`J-Link: Flash download: Bank 0 @ 0x00000000: Skipped. Contents already match`
`O.K.`
Which is extremely odd for me

Am I doing something wrong to add support for this board?

I'm attaching the edited files.

Thank you very much

[modified_files.zip](https://github.com/openthread/openthread/files/3734449/modified_files.zip)

nigelg-silabs, 2019-10-16:
I'm not sure if it is the fix but there was a channel range issue on the 4170 which is in PR.

Have a look at:
https://github.com/openthread/openthread/pull/4214

mselser, 2019-10-22:
Rookie mistake. Antennas were not connected. My bad.

closed 2019-10-22 by mselser