openthread/openthread #4262: Issue with UDP between border routers

embeddedRSA, 2019-10-22:
Hardware:
client: Nrf52840 using Ubuntu 18.04.3 LTS 
server: rpi boder router

SW: 
SW im using the docker image shown here https://openthread.io/guides/border-router/docker/run
and running a build with the flags: 
examples/Makefile-nrf52840 USB=1 BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1

issue:
i've been using iperf3 for testing bandwidth, when using tcp i get a connection , but when running UDP nothing happens

This is the output, which suggests that the UDP message is sent but not received.
`
root@b19acd84623e:/app/borderrouter# sudo iperf3 -c  fd11:22::d0b0:dd2f:b47b:fa18 -u -t 5 --get-server-output
Connecting to host fd11:22::d0b0:dd2f:b47b:fa18, port 5201
[  4] local fd11:22::4a73:c997:faa8:cbe3 port 58061 connected to fd11:22::d0b0:dd2f:b47b:fa18 port 5201
[ ID] Interval           Transfer     Bandwidth       Total Datagrams
[  4]   0.00-1.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   1.00-2.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   3.00-4.00   sec   128 KBytes  1.05 Mbits/sec  16  
[  4]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec  16  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Jitter    Lost/Total Datagrams
[  4]   0.00-5.00   sec   640 KBytes  1.05 Mbits/sec  0.000 ms  0/0 (0%)  
[  4] Sent 0 datagrams

Server output:
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from fd11:22::4a73:c997:faa8:cbe3, port 56058
[  5] local fd11:22::d0b0:dd2f:b47b:fa18 port 5201 connected to fd11:22::4a73:c997:faa8:cbe3 port 58061
[ ID] Interval           Transfer     Bandwidth       Jitter    Lost/Total Datagrams
[  5]   0.00-1.00   sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  
[  5]   6.00-6.58   sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Jitter    Lost/Total Datagrams
[  5]   0.00-6.58   sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  

iperf Done.
`

I've been trying to use tcpdump to identify whether UDP connection is running at all, I did see some packages on the server which would suggest yes, but im not sure, since the test results don't agree. 

**Client side as expected  (note that screenshot is only partial since the actual output is homogeneous in its entirety:**

root@b19acd84623e:/app/borderrouter# sudo tcpdump -i wpan0 udp
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wpan0, link-type RAW (Raw IP), capture size 262144 bytes
06:42:39.096680 IP6 fd11:22::4a73:c997:faa8:cbe3.58061 > fd11:22::d0b0:dd2f:b47b:fa18.5201: UDP, length 4
06:42:39.111370 IP6 fd11:22::d0b0:dd2f:b47b:fa18.5201 > fd11:22::4a73:c997:faa8:cbe3.58061: UDP, length 4
06:42:39.151538 IP6 fd11:22::4a73:c997:faa8:cbe3 > fd11:22::d0b0:dd2f:b47b:fa18: frag (0|1232) 58061 > 5201: UDP, bad length 8192 > 1224
06:42:39.151567 IP6 fd11:22::4a73:c997:faa8:cbe3 > fd11:22::d0b0:dd2f:b47b:fa18: frag (1232|1232)
06:42:39.151582 IP6 fd11:22::4a73:c997:faa8:cbe3 > fd11:22::d0b0:dd2f:b47b:fa18: frag (2464|1232)
06:42:39.151584 IP6 fd11:22::4a73:c997:faa8:cbe3 > fd11:22::d0b0:dd2f:b47b:fa18: frag (3696|1232)
06:42:39.151585 IP6 fd11:22::4a73:c997:faa8:cbe3 > fd11:22::d0b0:dd2f:b47b:fa18: frag (4928|1232)
06:42:39.151587 IP6 fd11:22::4a73:c997:faa8:cbe3 > fd11:22::d0b0:dd2f:b47b:fa18: frag (6160|1232)
06:42:39.151588 IP6 fd11:22::4a73:c997:faa8:cbe3 > fd11:22::d0b0:dd2f:b47b:fa18: frag (7392|808)
06:42:39.251698 IP6 fd11:22::4a73:c997:faa8:cbe3 > fd11:22::d0b0:dd2f:b47b:fa18: frag (0|1232) 58061 > 5201: UDP, bad length 8192 > 1224
06:42:39.251717 IP6 fd11:22::4a73:c997:faa8:cbe3 > fd11:22::d0b0:dd2f:b47b:fa18: frag (1232|1232)

**The server output on the other hand seems to receive only a few packets:**

pi@raspberrypi:~ $ sudo tcpdump -i wpan0 udp
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wpan0, link-type RAW (Raw IP), capture size 262144 bytes
06:42:39.373550 IP6 fd11:22::4a73:c997:faa8:cbe3.58061 > fd11:22::d0b0:dd2f:b47b:fa18.5201: UDP, length 4
06:42:39.373894 IP6 fd11:22::d0b0:dd2f:b47b:fa18.5201 > fd11:22::4a73:c997:faa8:cbe3.58061: UDP, length 4
^C
2 packets captured
2 packets received by filter
0 packets dropped by kernel

Does anyone have a fix? i would expect UDP to be possible, i might have forgotten something in my setup.

jwhui, 2019-10-22:
@embeddedRSA, a primary difference between TCP and UDP is that the former provides end-to-end flow and congestion control. When testing with UDP, I believe you are simply overwhelming the network and success rate is very poor.

1. The UDP packet size is larger than Thread's IPv6 MTU, which means IPv6 fragmentation is 
enabled.

1. Each IPv6 fragment is 1232 bytes, which means 6LoWPAN fragmentation is enabled to deliver large IPv6 datagrams.

1. The data rate is configured for 1.05 Mbits/sec, which is far higher than what Thread can support - the IEEE 802.15.4 PHY has a raw data rate of supports 250 Kbits/sec.

I suggest starting with a lower target bandwidth (`-b` option) and message size (`-l` option).

Hope that helps.

embeddedRSA, 2019-10-23:
Thanks alot! adding the two flags and the parameters -b 100k and -l 1000 seem to work.
so yes it helped!

closed 2019-10-23 by embeddedRSA