openthread/openthread #4271: [nrf528xx] make pretty problem

Irving-cl, 2019-10-24:
**Describe the bug**
When I run 'make pretty', I got the following error:

```
Making pretty in nrf528xx
make[3]: *** No rule to make target 'src/*.c', needed by 'pretty'. Stop.
make[2]: *** [/usr/local/google/home/irvingcl/Documents/git/openthread-1.2/examples/../third_party/nlbuild-autotools/repo/automake/post/rules/pretty.am:69: pretty-recursive] Error 1
make[1]: *** [/usr/local/google/home/irvingcl/Documents/git/openthread-1.2/examples/../third_party/nlbuild-autotools/repo/automake/post/rules/pretty.am:69: pretty-recursive] Error 1
```

It seems that wildcard cannot be used here. It take src/*.c as a real file.

**To Reproduce**
1.    I ran it on a x86_64 Linux machine.
2.    After going to the root directory of the repo, run `./bootstrap`.
3.    Then make: `make -f examples/Makefile-posix`
4.    Next make-pretty: `make -C build/x86_64-unknown-linux-gnu pretty`

And Then I met this problem.

@konradderda

konradderda, 2019-10-24:
I have created a PR that resolves this problem: #4272
Issue does not occur if you call './configure' and then 'make pretty' from the root directory. Anyway, usage of Makefile wildcard function seems to resolve the problem.

jwhui, 2019-10-25:
Resolved by #4272.

closed 2019-10-25 by jwhui