openthread/openthread #4284: [spinel] how to reset the counters

LuDuda, 2019-10-29:
@abtink We would like to implement a mechanism to clear all counters i.e. MAC Counters, MLE Counters, and IP counters via spinel and also via CLI.

I spotted that there is currently following SPINEL properties (not implemented):
```
    /// Counter reset behavior
    /** Format: `C`
     *  Writing a '1' to this property will reset
     *  all of the counters to zero. */
    SPINEL_PROP_CNTR_RESET = SPINEL_PROP_CNTR__BEGIN + 0,
```

I just want to ensure, that writing "1" to this property should reset ALL (MAC, MLE, IP) counters in the OT? If yes, then clearing MAC, MLE or IP counters is fine, but one strange thing is that under `PROP_CNTR` space I do also see `SPINEL_PROP_MSG_BUFFER_COUNTERS` which basically can't be cleared.

Some alternative is to provide new properties for all counters type which will clear them individually, or make use of the `SPINEL_CMD_PROP_VALUE_SET` command (instead of `SPINEL_CMD_PROP_VALUE_GET`) which could clear the counters when the payload is empty/0.

abtink, 2019-10-29:
@LuDuda , I think it's a good idea to use `VALUE_SET` (with any value not necessarily `1`) to  `SPINEL_PROP_CNTR_RESET` to clear all counters (MAC, MLE and IP) that can be cleared.

As you point out the `SPINEL_PROP_MSG_BUFFER_COUNTERS` is different in the sense that it's the current number of msg/data buffers in use by all msg queues and we cannot (should not) clear it.

If we want more control, we can also implement `VALUE_SET` to `SPINEL_PROP_CNTR_ALL_MAC_COUNTERS` or `SPINEL_PROP_CNTR_MLE_COUNTERS` to only clear MAC or MLE counters.

I can help add this later.

LuDuda, 2019-10-29:
Hello @abtink if you have time and a clear vision on how to implement it and document I would appreciate the help! In meantime I've added [pyspinel](https://github.com/openthread/pyspinel/pull/83) support for retrieving counters, later I can extend it with resetting mechanism (same with CLI).

abtink, 2019-10-30:
@LuDuda Sounds good! Just submitted PR https://github.com/openthread/openthread/pull/4286.

jwhui, 2019-11-01:
Resolved by #4286.

closed 2019-11-01 by jwhui