openthread/openthread #4290: Getting Started: Developing with OpenThread APIs build errors

Tobamalau, 2019-10-30:
Hello,

I followed the Getting Started: Developing with OpenThread APIs to point 11 where I have to build OpenThread (make -f examples/Makefile-nrf52840). Here I get some errors

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id: efd9ebaa
2. hardware platform nrf52840
3. Build steps like getting started

**Console output in step 
           Making all in cli 
           CC      ot_cli_mtd-main.o**

/examples/apps/cli/main.c:58:13: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'handleNetifStateChanged'
 void OTCALL handleNetifStateChanged(uint32_t aFlags, void *aContext);

/examples/apps/cli/main.c:143:41: error: 'handleNetifStateChanged' undeclared (first use in this function)

/examples/apps/cli/main.c:287:31: error: 'OT_NETIF_INTERFACE_ID_THREAD' undeclared (first use in this function)

/examples/apps/cli/main.c:308:17: error: 'otMessageInfo {aka struct otMessageInfo}' has no member named 'mInterfaceId'; did you mean 'mIsHostInterface'?

/examples/apps/cli/main.c: At top level:
cc1: error: unrecognized command line option '-Wno-gnu-zero-variadic-macro-arguments' [-Werror]

are there any tips for this errors?

Vyrastas, 2019-10-30:
@Tobamalau can you double-check your main.c code?  You most likely are missing a semicolon after a statement somewhere.

Vyrastas, 2019-10-30:
Actually, running through the Codelab I encountered the same issue.  It appears the issue is caused by the removal of the otNetifInterfaceId enumeration in #3925 (of which OT_NETIF_INTERFACE_ID_THREAD was an element)...

@jwhui What should we use instead?  It's in a couple places here https://codelabs.developers.google.com/codelabs/openthread-apis/#6

jwhui, 2019-10-30:
You should be able to just remove the following line from the example code:
```c++
listenSockAddr.mScopeId = OT_NETIF_INTERFACE_ID_THREAD;
```
#3925 has many similar examples.

Vyrastas, 2019-10-30:
Still got errors after that, caused by the removal of OTCALL in #3878 ... removing those finally fixed it.

@Tobamalau you'll need to do this:

In main.c, remove these lines:
`listenSockAddr.mScopeId = OT_NETIF_INTERFACE_ID_THREAD;`
`messageInfo.mInterfaceId = OT_NETIF_INTERFACE_ID_THREAD;`

Also in main.c, delete `OTCALL` from the `handleNetifStateChanged` function declaration and definition.

After that it should build.  I'll update the Codelab, thanks for bringing this to our attention.

Tobamalau, 2019-10-31:
it works, thanks a lot.
The tutorials are very helpfully!

Vyrastas, 2019-10-31:
The codelab has been updated.
https://codelabs.developers.google.com/codelabs/openthread-apis/#0

closed 2019-10-31 by Vyrastas