openthread/openthread #4301: non volatile persist data not saved after join

musalk-dk, 2019-11-04:
**Describe the bug**
After joining at network the persistent non volatile data is not stored correctly. And the non volatile data is invalid. 

**To Reproduce**
code setup: We have our network join/commissioning as part of our bootloader selected via a key press'n hold on power on.
So the join process is part of a pre bootloader state (to ensure we can always bootload). Our openthread persist data is shared by the bootloader and the application. If only commissioning is requested the pre bootloader will after a successful join, reset to enter application with a clean slate.
We used otPlatReset to do the reset.
We also tried to do a otThreadSetEnable = false and otInstanceFinalize before the reset.

Set thread to join a network (works) reset
1. Git commit id: 2279ef610eef80597bd789611d6931b95e08bf48
2. IEEE 802.15.4 hardware platform: nrf52840
3. Build steps: Nordic semi prebuilt lib's v3.1.0
4. Network topology RPi gateway/commissioner, 2 FTD nodes

**Expected behavior**
We expected that otPlatReset would ensure all writes are done before the reset or at least in combination with thread disable and instance finalize as we have found no other api call's to ensure writes to non volatile are done.

**Console/log output**
N/A

**Additional context**
Maybe they exists and we have not found them but we really need 2 api functions for the non volatile storage part.
One that tells thread that power out is imminent and to finalize non volatile writes.
and one to check if writes are done before software induced resets.
This is quite critical as we have already seen the flash corruption even outside the thread data area on multiple occasions. (nrf52840 has fixed brownout enabled).
This is properly a chip issue but having some api calls to help do a workaround and to ensure persist data is saved before both un- and intentional resets would in any case be a plus.

musalk-dk, 2019-11-04:
We have a semi working version where we delay the reset by 10sec (1sec was not enough). But that's a bad workaround as we 
A. have a properly unnecessary long user interface delay.
B. we have no insurance that thread actually did the writes within the that time.

jwhui, 2019-11-04:
OpenThread does not currently save network parameters obtained via Thread Commissioning until the device successfully attaches to an existing Thread network (e.g. by becoming a end-device or router/leader). In your scenario above, is the device successfully attaching to the network? Or is that not a part of the flow that you are trying to support?

musalk-dk, 2019-11-04:
We get the device to successfully commission when we continue into our bootloader maintaining the same instance... But when we only need to use the bootloader to commission and then reset to application - the otherwise successful join is lost. Most likely due to data not written before it becomes some kind of node. 
So if we wait for it to become something else than detached it should have saved the connection data?
Will any of the disable/reset/finalize functions ensure all flash writes are done? Before eg. A soft reset.

musalk-dk, 2019-11-08:
Will any of the disable/reset/finalize functions ensure all flash writes are done? 
Before eg. A soft reset.?

jwhui, 2019-11-12:
> * Will any of the disable/reset/finalize functions ensure all flash writes are done before eg. A soft reset.?

Not currently. We could look to an enhancement that saves information received as part of Thread Commissioning to non-volatile.

> * Is thread_process required to be called after a disable/finalize to do actions part of the shutdown?

No, once `otInstanceFinalize()` is called, the OpenThread instance object is no longer valid. (Note that `thread_process()` is specific to the Nordic SDK).

jwhui, 2019-11-14:
@musalk-dk, I just submitted #4332, which should store the Thread network parameters as soon as they are received. Let me know if #4332 addresses your use case.

jwhui, 2019-11-25:
Resolved by #4332.

closed 2019-11-25 by jwhui