openthread/openthread #4316: OT BR documentation question

otbr-user, 2019-11-10:
When running OT BR with docker is it necessary to explicitly add static routes and bridge interfaces?

According to this https://openthread.io/guides/border-router/docker/test-connectivity simply starting the container is enough to get IPv6 connectivity between the Thread network and public internet. Is this really true though? How would the On-Mesh prefix be known to the host OS?

jwhui, 2019-11-12:
Starting the OTBR docker container provides public Internet connectivity, but not necessarily IPv6 connectivity. Note that the referenced page describes connectivity using NAT64. As a result, it is not necessary for the container to have a publicly routable IPv6 address or prefix. For direct IPv6 connectivity to the public Internet, the container needs to have publicly routable IPv6 address/prefix.

Connectivity between host OS and Thread Network is a different scenario from public Internet connectivity. If you are looking for connectivity with host OS, then static routes and/or bridging would be appropriate.

otbr-user, 2019-11-12:
That page states 
> You should be able to a ping a public IPv6 address from the standalone Thread node at this point.
Without additional configuration the container does not have a publicly routable IPv6 address, right? So the statement quoted above is not correct?

jwhui, 2019-11-14:
Yes, it appears the documentation is incorrect. The example demonstrates how to ping public IPv4 addresses using NAT64.

Vyrastas, 2019-11-14:
This has been fixed:
https://openthread.io/guides/border-router/docker/test-connectivity#ping-a-public-address

Thanks for bringing it up @otbr-user

closed 2019-11-14 by Vyrastas