openthread/openthread #4327: Thread 1.2.0 Core Implementation

jwhui, 2019-11-13:
This issue tracks Thread 1.2.0 core feature implementation.

- [ ] Thread 1.2 feature flag (#4325)
- [ ] Enhanced Keep Alive (#4325)
- [ ] Enhanced Frame Pending (TBD)
- [ ] Link Metrics (TBD)
- [ ] Coordinated Sampled Listening (TBD)
- [ ] Domain Unicast Address Registration (TBD)
- [ ] Multicast Address Registration (TBD)

zhifeng1756, 2020-10-22:
Does Openthread supports Commissioning and configuration via BLE as stated in Thread [v1.2](https://www.threadgroup.org/Portals/0/documents/support/ThreadInCommercialWhitePaper_2542_1.pdf)? If not yet, any plan?

Thanks,

jwhui, 2020-10-22:
> Does Openthread supports Commissioning and configuration via BLE as stated in Thread [v1.2](https://www.threadgroup.org/Portals/0/documents/support/ThreadInCommercialWhitePaper_2542_1.pdf)? If not yet, any plan?

@zhifeng1756 , OpenThread does not currently support Bluetooth Extensions. Currently, we do not have a roadmap to support this feature.

khaled34, 2021-03-17:
What will be the release date for openthread 1.2 ?

closed 2021-05-15 by jwhui