openthread/openthread #4342: Joiner fails after forming network or join to network

al1img, 2019-11-19:
Once the joiner device forms the network it can't join to another network after clearing the active dataset.

Git commit id: 9f2ff3bcbfd837adbc9c6280c48fbefd01d3bd01
IEEE 802.15.4 hardware platform: posix, efr32, probably all other
Build steps: `make -f examples/Makefile-posix JOINER=1 COMMISSIONER=1`

Steps to reproduce:

1. Start the joiner device and form the network:
```
> dataset init new
Done
> dataset commit active
Done
> ifconfig up
Done
> thread start
Done
> state
leader
Done
```
2. Stop the network and clear active dataset:
```
> thread stop
Done
> ifconfig down
Done
> dataset clear
Done
> dataset commit active
Done
```
3. Start the commissioner device, form network and add joiner:
```
> dataset init new
Done
> dataset commit active
Done
> ifconfig up 
Done
> thread start
Done
> commissioner start
Commissioner: petitioning
Done
> Commissioner: active
> commissioner joiner add * J01NME
Done
```
4. Start joiner on the joiner device:
```
> ifconfig up
Done
> joiner start J01NME
Done
> Join failed [NotFound]
```
I expect that the joiner device is joined successfully but it fails with `NotFound` error. 
It joins successfully only after resetting the joiner.

jwhui, 2019-11-19:
@al1img, I believe this is currently due to the PAN ID value being cached. Can you try setting the PAN ID to `0xffff` before joining?

Note that the Joiner will use the PAN ID setting as a filter in its scan. If the PAN ID is not `0xffff`, then only networks with the matching PAN ID will be discovered.

al1img, 2019-11-20:
Setting PAN ID to 0xffff resolves the issue. So, if the joiner needs to be joined to another network, setting PAN ID to 0xffff should be enough.

closed 2019-11-25 by al1img

PathToLife, 2025-02-26:
Google redirected me here.

For anyone also using the open thread cli (possibly on a esp32), the command to set panid to 0xffff is

`> panid 0xffff`

Thank you for the solution jwhui