openthread/openthread #4346: [ncp] logging enhancements

LuDuda, 2019-11-20:
We would like to work on some logging enhancements in NCP architecture. But first, I want to ensure this is something desirable for the community, and technically acceptable.

1. Dynamic log level configuration per module/region.

Currently, OpenThread supports dynamic configuration of the log level, this log level is the same for each enabled log region. We would like to introduce the option to dynamically change the log level for each region, via SPINEL (or compile-time). Because each region has it's own macros used for logging this seems to be quite easy to implement and may add some flash/ram only if special define would be enabled.

2. Timestamping NCP logs.

Because logs come asynchronously from NCP to Host, sometimes it's hard to debug timing-related issues. Wpantund/syslog timestamps logs based on the time when the SPINEL frames are received by the Host, and not when there were generated on the NCP. The proposal is to add another field to SPINEL_PROP_STREAM_LOG property, which will be the timestamp (64-bit value - number of ms from the NCP boot). As an improvement, the Host side could send the current Epoch time, so that NCP could timestamp the log based on the configured Epoch time. Of course, this will not allow for perfect synchronization (there will be always some lag between sending the property from Host side to receiving it on NCP side) but will help with debugging.

jwhui, 2019-11-21:
@LuDuda both proposed enhancements sound good to me!

abtink, 2019-11-21:
I agree. 
 For the `SPINEL_PROP_STREAM_LOG` we should be able to easily add more optional fields in the meta data part.

konradderda, 2020-01-01:
@LuDuda @jwhui  @abtink I have created PR that resolves 2nd point:
https://github.com/openthread/openthread/pull/4439

closed 2024-10-12 by LuDuda