openthread/openthread #4353: [efr32] Parent does not purge indirect message

NickNamerr, 2019-11-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id 5c645c68c1138fa56e4a1f45f3b496c4b4a4462b
2. IEEE 802.15.4 hardware platform EFR32MG12
3. Build steps -f examples/Makefile-efr32mg12 SERVICE=1 COAP=1 BOARD=BRD4172A UDP_FORWARD=1 LINK_RAW=1
4. Network topology 1 SED, 1 REED, 1BR

**Expected behavior**
Parent will purge transmitted indirect messages.

**Console/log output**
Does not applicable

**Additional context**
During test of INDIRECT messages noticed that parent is needed ~100ms to decide that indirect message could be cleared. If do soft reset on SED within 10 to 40 ms after indirect messages been received, after boot up this messages will be received one more time (could be repeated multiple times). Is there any API to get informed about that all indirect messages been delivered to SED and Parent does not keep anymore indirect messages? I tried to call "instance.Get<DataPollSender>().StopFastPolls();" but seems like does not work.

jwhui, 2019-11-26:
@NickNamerr, thank you for raising this issue.

Can you provide more information on the indirect message? Are the unicast or multicast? What IPv6 destination address(es) are in the packets? Where are the packets being originated from?

NickNamerr, 2019-11-27:
@jwhui this is unicast COAP message sends by node.js CoapServer to SED local-mesh IPv6 (based on Extended Panid). 
If implement a code with logic to call otPlatReset after SED receives any CoAP message on dedicated CoAP uri, it will be in recursion of receiving same message until parent will not purge it by timeout. That something expected because device does not send confirmation of receiving indirect message. If add scheduling of reset for ~20-40ms  i my case device could receive same message up to 3-5 times.

jwhui, 2019-11-27:
@NickNamerr, thanks for the additional info. Can you provide a packet capture using a sniffer?

NickNamerr, 2019-11-27:
@jwhui 
[image]
I cannot provide pcap data, because it consist private data, but can provide screenshot and explain it:
::fc2c:7c45:f5db:ef38 - wpan0 IPv6
::dfd6:94fa:bb86:748f - Router IPv6
fe80::dcb7:24dd:5ef:8e85 - SED IPv6
fe80::b8b3:d4c6:934c:99b5 - Router IPv6
0x9400 - Router RLOC16
0x9401 - SED RLOC16
/ka - CoAP topic
Scenario:
1) SED is in EM4 mode;
2) I am sending packer to SED , that decoding it and calling otPlatReset in 10ms after receiving it;
3) SED doing reset and receiving same packet again and again.

NickNamerr, 2019-11-27:
@jwhui btw:
[image]
Why sniffer shows same message was sent 4 times? My tcpdump on server shows only one in and one out messages.

jwhui, 2019-11-27:
@NickNamerr, thanks for sharing the packet trace.

I see two CoAP requests getting delivered to the SED (packets 320 and 347). I do not see an ACK to 320, which can be due to sniffer dropping the packet or ACK is not getting delivered to parent.

Can you determine from the OpenThread logs whether an ACK is being received on the parent in response to the CoAP request messages?

NickNamerr, 2019-11-28:
Actually I do not have time to debug it now. I close this ticket and reopen in future once will have enough time to debug.
Thanks and sorry.

closed 2019-11-28 by NickNamerr