openthread/openthread #4381: Router path is 0 when only 2 devices in network

musalk-dk, 2019-12-05:
**Describe the bug**
Showing the router table - shows the opposing unit with a with a path cost of 0 when only 2 devices are in network.
When a 3rd unit is added the path costs correctly shows 1 for the opposing units.

**To Reproduce**
Information to reproduce the behavior, including:
Git commit id: 2279ef6
IEEE 802.15.4 hardware platform: nrf52840
Build steps: Nordic semi prebuilt lib's v3.1.0
Network topology RPi gateway/commissioner, 1/2 FTD nodes

**Expected behavior**
Path cost of a neighboring unit should always be at least 1.

**Console/log output**
N/A

**Additional context**
N/A

jwhui, 2019-12-05:
This is expected behavior.

The router table really stores two different ways of reaching the device:
1. Direct link - represented by LQ In/Out
2. Multihop path - represented by Next Hop and Path Cost.

The multihop path is only valid if there is a path that requires 2 or more hops to reach the router. In a topology with only 2 devices, no such paths exist.

The reason for storing direct link and multihop path separately is it gives an immediate failover option to a multihop path if the direct link fails.

musalk-dk, 2019-12-10:
Thanks.
The confusion comes when more than 3 devices are all in reach (direct link) the path becomes 1 for the opposing units all cases I have seen. But when only 2 the path becomes 0.
Thanks again

closed 2019-12-10 by musalk-dk