openthread/openthread #4393: [Settings] When using OPENTHREAD_SETTINGS_RAM=1 can`t restore devices ID manualy after reboot

AlexeyShako, 2019-12-11:
Hi! 
    I was faced with issue, which forbids me to start a network with my previously known parameters (saved manually before platform reset) when I was using OPENTHREAD_SETTINGS_RAM=1 for NRF52840. I want manually fill the volatile memory buffer with my parameters before the start, but I can`t because it will be re-erased during the next init steps. 
   I figured out of that by covering Get<Settings>().Init(); with preprocessor directives. This method requires manual call otPlatSettingsInit() but after that my predefined configs could be added to the volatile storage and be found by internal open thread calls to the volatile storage. In my opinion, it will be great to have this feature in the official repo.

Part of updated code:
`
void Instance::AfterInit(void)
{
    mIsInitialized = true;
#if OPENTHREAD_MTD || OPENTHREAD_FTD
    // Restore datasets and network information
#if !OPENTHREAD_SETTINGS_RAM
    Get<Settings>().Init();
#endif
    Get<Mle::MleRouter>().Restore();
#endif // OPENTHREAD_MTD || OPENTHREAD_FTD
#if OPENTHREAD_ENABLE_VENDOR_EXTENSION
    Get<Extension::ExtensionBase>().SignalInstanceInit();
#endif
}
`

jwhui, 2019-12-11:
@AlexeyShako, thank you for submitting this issue.

Can you clarify exactly what information you are trying to restore? Is it the IEEE Extended Address? Something else?

For IEEE Extended Address, is it possible to use the existing [otLinkSetExtendedAddress()](https://openthread.io/reference/group/api-link-link#otlinksetextendedaddress) API?

AlexeyShako, 2019-12-11:
I'm trying to have constant node parameters while using a volatile memory. I expected completely the same behavior as with using flash storage, after network establishment. But I've found API for getting/setting only global parameters of the network. I put manually saved dataset to otDatasetSetActive() and network successfully starts, but all identifiers didn't constant. That's exactly the issue for my application. The goal of this task to keep all nodes with the same ID in the network even after node reset.
I find that all required parameters placed in the volatile buffer with key "kKeyNetworkInfo". And after saving it locally and restoring with help  proposed above tweak - I got the same result as expected
Has my goal become clearer?

By the way if I always set the same IEEE Extended Address, will the device "mliid" parameter in network info structure stay constant?

jwhui, 2019-12-11:
Given that your proposal is to effectively remove the call to `otPlatSettingsInit()`, can your platform simply provide an alternative implementation of `otPlatSettingsInit()` that populates the information instead of clearing it?

Note that one issue without non-volatile memory is managing the frame counters used with AES-CCM message security. The frame counter is included in the nonce and, as such, must only be used once for a given AES key and extended address. Another issue is that replay projection is based on frame counters. If frame counters are reset, neighbors with an established link will drop the message until the link is re-established.

AlexeyShako, 2019-12-16:
Got it, thanks for your clarification. Could you please recommend any correct approach on how to keep some constant identification for each node in the network since it joined the network? Because I need to create some logical connection over the network related to specified nodes, but I can`t since it always changes its logical address and I can`t predict when it changes (I mean case when a node reboots). Of course, it`s using volatile memory. When I`m using FLASH memory, everything works great.

jwhui, 2019-12-16:
One possible approach is to manually assign your own IPv6 addresses to each of the nodes using [otIp6AddUnicastAddress](https://openthread.io/reference/group/api-ip6#otip6addunicastaddress). When a node reboots, they will randomly select new identifiers for IEEE 802.15.4 Extended Address and Mesh Local Endpoint Identifier. However, you can manually assign your own IPv6 addresses.

Note: for the manually assigned IPv6 addresses to work, you will need to register an on-mesh prefix so that devices know to perform IPv6 address discovery within the Thread network. See [otBorderRouterAddOnMeshPrefix](https://openthread.io/reference/group/api-border-router#otborderrouteraddonmeshprefix). Or if you are using `wpantund`, you can reference the `wpanctl add-prefix` command.

jwhui, 2020-02-10:
Closing stale issue.

closed 2020-02-10 by jwhui