openthread/openthread #4395: Compilation Error when OPENTHREAD_CONFIG_MLE_MAX_CHILDREN is 511

jsnayak, 2019-12-12:
I am facing a compilation issue when the OPENTHREAD_CONFIG_MLE_MAX_CHILDREN is set to 511

**To Reproduce**
Information to reproduce the behavior, including:
1. fb14009916c39db43bf637648b0ff3c17e454f5f
2. kw41z
3. Change the configuration parameter to 512 and build, I used the below build flags:
`make -f examples/Makefile-kw41z BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 JOINER=1 UDP_PROXY=1 UDP_FORWARD=1 COAP=1 FULL_LOGS=1`

**Console/log output**
`  CXX      api/libopenthread_mtd_a-diags_api.o
In file included from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/common/code_utils.hpp:41:0,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp:43,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/api/coap_secure_api.cpp:39:
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp: In member function 'const ot::Coap::Message::HelpData& ot::Coap::Message::GetHelpData() const':
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/utils/static_assert.hpp:69:105: error: invalid use of incomplete type 'struct ot::StaticAssertError<0>'
         _OT_STATIC_ASSERT_ERROR(__COUNTER__) = sizeof(ot::StaticAssertError<(aExpression) != 0>(aMessage)), \
                                                                                                         ^
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp:598:9: note: in expansion of macro 'OT_STATIC_ASSERT'
         OT_STATIC_ASSERT(sizeof(mBuffer.mHead.mInfo) + sizeof(HelpData) + kHelpDataAlignment <= sizeof(mBuffer),
         ^~~~~~~~~~~~~~~~
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/utils/static_assert.hpp:48:23: note: declaration of 'struct ot::StaticAssertError<0>'
 template <int> struct StaticAssertError;
                       ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/common/code_utils.hpp:41:0,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp:43,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap.hpp:36,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/common/instance.hpp:54,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/api/child_supervision_api.cpp:37:
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp: In member function 'const ot::Coap::Message::HelpData& ot::Coap::Message::GetHelpData() const':
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/utils/static_assert.hpp:69:105: error: invalid use of incomplete type 'struct ot::StaticAssertError<0>'
         _OT_STATIC_ASSERT_ERROR(__COUNTER__) = sizeof(ot::StaticAssertError<(aExpression) != 0>(aMessage)), \
                                                                                                         ^
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp:598:9: note: in expansion of macro 'OT_STATIC_ASSERT'
         OT_STATIC_ASSERT(sizeof(mBuffer.mHead.mInfo) + sizeof(HelpData) + kHelpDataAlignment <= sizeof(mBuffer),
         ^~~~~~~~~~~~~~~~
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/utils/static_assert.hpp:48:23: note: declaration of 'struct ot::StaticAssertError<0>'
 template <int> struct StaticAssertError;
                       ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/common/code_utils.hpp:41:0,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp:43,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap.hpp:36,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/common/instance.hpp:54,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/api/diags_api.cpp:38:
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp: In member function 'const ot::Coap::Message::HelpData& ot::Coap::Message::GetHelpData() const':
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/utils/static_assert.hpp:69:105: error: invalid use of incomplete type 'struct ot::StaticAssertError<0>'
         _OT_STATIC_ASSERT_ERROR(__COUNTER__) = sizeof(ot::StaticAssertError<(aExpression) != 0>(aMessage)), \
                                                                                                         ^
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp:598:9: note: in expansion of macro 'OT_STATIC_ASSERT'
         OT_STATIC_ASSERT(sizeof(mBuffer.mHead.mInfo) + sizeof(HelpData) + kHelpDataAlignment <= sizeof(mBuffer),
         ^~~~~~~~~~~~~~~~
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/utils/static_assert.hpp:48:23: note: declaration of 'struct ot::StaticAssertError<0>'
 template <int> struct StaticAssertError;
                       ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/common/code_utils.hpp:41:0,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp:43,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/api/coap_api.cpp:38:
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp: In member function 'const ot::Coap::Message::HelpData& ot::Coap::Message::GetHelpData() const':
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/utils/static_assert.hpp:69:105: error: invalid use of incomplete type 'struct ot::StaticAssertError<0>'
         _OT_STATIC_ASSERT_ERROR(__COUNTER__) = sizeof(ot::StaticAssertError<(aExpression) != 0>(aMessage)), \
                                                                                                         ^
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp:598:9: note: in expansion of macro 'OT_STATIC_ASSERT'
         OT_STATIC_ASSERT(sizeof(mBuffer.mHead.mInfo) + sizeof(HelpData) + kHelpDataAlignment <= sizeof(mBuffer),
         ^~~~~~~~~~~~~~~~
/home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/utils/static_assert.hpp:48:23: note: declaration of 'struct ot::StaticAssertError<0>'
 template <int> struct StaticAssertError;
                       ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/common/code_utils.hpp:41:0,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap_message.hpp:43,
                 from /home/ubuntu/src/git_code/ot-ncp/OT_fb140/examples/../src/core/coap/coap.hpp:36,
[... 88 lines cut ...]
make[4]: *** [api/libopenthread_mtd_a-coap_secure_api.o] Error 1
Makefile:4186: recipe for target 'api/libopenthread_mtd_a-child_supervision_api.o' failed
make[4]: *** [api/libopenthread_mtd_a-child_supervision_api.o] Error 1
Makefile:4172: recipe for target 'api/libopenthread_mtd_a-channel_monitor_api.o' failed
make[4]: *** [api/libopenthread_mtd_a-channel_monitor_api.o] Error 1
Makefile:4256: recipe for target 'api/libopenthread_mtd_a-dataset_api.o' failed
make[4]: *** [api/libopenthread_mtd_a-dataset_api.o] Error 1
Makefile:4284: recipe for target 'api/libopenthread_mtd_a-diags_api.o' failed
make[4]: *** [api/libopenthread_mtd_a-diags_api.o] Error 1
Makefile:4228: recipe for target 'api/libopenthread_mtd_a-commissioner_api.o' failed
make[4]: *** [api/libopenthread_mtd_a-commissioner_api.o] Error 1
Makefile:4242: recipe for target 'api/libopenthread_mtd_a-crypto_api.o' failed
make[4]: *** [api/libopenthread_mtd_a-crypto_api.o] Error 1
Makefile:4158: recipe for target 'api/libopenthread_mtd_a-channel_manager_api.o' failed
make[4]: *** [api/libopenthread_mtd_a-channel_manager_api.o] Error 1
Makefile:4144: recipe for target 'api/libopenthread_mtd_a-border_router_api.o' failed
make[4]: *** [api/libopenthread_mtd_a-border_router_api.o] Error 1
Makefile:447: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:601: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
Makefile:527: recipe for target 'all' failed
make[1]: *** [all] Error 2
examples/Makefile-kw41z:226: recipe for target 'do-build-cortex-m0plus' failed
make: *** [do-build-cortex-m0plus] Error 2`

jwhui, 2019-12-12:
@jsnayak the error you are seeing is due to the message buffer size not being large enough to support the desired number of children. You could resolve this specified error by increasing [OPENTHREAD_CONFIG_MESSAGE_BUFFER_SIZE](https://github.com/openthread/openthread/blob/ab89f403850bfe5fc153a4f5cb5b38ae68800ea6/src/core/config/openthread-core-default-config.h#L181).

jsnayak, 2019-12-13:
@jwhui I updated the parameter [OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS](https://github.com/openthread/openthread/blob/ab89f403850bfe5fc153a4f5cb5b38ae68800ea6/src/core/config/openthread-core-default-config.h#L165) and saw the issue resolved. Thanks!

closed 2019-12-13 by jsnayak