openthread/openthread #4406: [build] output build configuration values

bukepo, 2019-12-13:
**Is your feature request related to a problem? Please describe.**
It will be very helpful if the build system could output the a .h file which shows all OpenThread build option values, especial when integrating OpenThread onto some platforms has very complicated build systems.

**Describe the solution you'd like**
I don't have a concrete solution yet.

**Describe alternatives you've considered**
Provide a debug version which shows all configuration values at runtime.

**Additional context**
There are multiple ways affecting OpenThread's configuration values. CPPFLAGS environment, project configuration files, OpenThread's default values, and automake generated files.

simonlingoogle, 2019-12-17:
```bash
simonlin@simonlin:~/Downloads$ cat test.h 
#define HELLO_HEADER 10
simonlin@simonlin:~/Downloads$ cat test.c
#include "test.h"
simonlin@simonlin:~/Downloads$ gcc -E test.c -dM | grep HELLO
#define HELLO_HEADER 10
```

`gcc -E -dM` can dump all macro defines. We need a specific cpp/hpp file for running `gcc -E -dM` on it with the same CPPFLAGS of other cpp files.

bukepo, 2019-12-17:
Great! @jwhui shall we export the effective core config header file to OpenThread users? That is install it as other OpenThread API header files.

bukepo, 2019-12-19:
friendly ping. @jwhui @abtink

abtink, 2019-12-19:
This seems related to: https://github.com/openthread/openthread/issues/4333

bukepo, 2019-12-19:
Got it! I'm find with keeping this generated header file private, just for debugging purpose. @abtink

abtink, 2019-12-19:
Having it for debugging sounds good and useful to me.

EskoDijk, 2025-07-17:
@bukepo Should this issue be assigned to someone else, or otherwise, closed, or unassigned?