openthread/openthread #4433: NetworkTime overflows on NRF52840

AlexeyShako, 2019-12-30:
Hi!
I'm using the latest version on NRF52840. And when I had tried to enable network time module using TIME_SYNC=1 flag, I found in logs that time moved for a years in a moment. Other options used for libraries build is next:
```
OPEN_THREAD_CONFIG += BORDER_AGENT=1
OPEN_THREAD_CONFIG += BORDER_ROUTER=1
OPEN_THREAD_CONFIG += COAP=1
OPEN_THREAD_CONFIG += COMMISSIONER=1
OPEN_THREAD_CONFIG += DNS_CLIENT=1
OPEN_THREAD_CONFIG += JOINER=1
OPEN_THREAD_CONFIG += LINK_RAW=1
OPEN_THREAD_CONFIG += MAC_FILTER=1
OPEN_THREAD_CONFIG += MTD_NETDIAG=1
OPEN_THREAD_CONFIG += SERVICE=1
OPEN_THREAD_CONFIG += UDP_FORWARD=1
OPEN_THREAD_CONFIG += ECDSA=1
OPEN_THREAD_CONFIG += SNTP_CLIENT=1
OPEN_THREAD_CONFIG += COAPS=1
OPEN_THREAD_CONFIG += FULL_LOGS=1
OPEN_THREAD_CONFIG += BOOTLOADER=BLE
OPEN_THREAD_CONFIG += DISABLE_BUILTIN_MBEDTLS=0
OPEN_THREAD_CONFIG += SETTINGS_RAM=1
```
And I`m attaching a small part of logs. It might be helpful:
```
THREAD|17313.028|15 CRIT [CRIT]-CORE----: NetworkTime: TimeSync::GetTime: offset:20010C3C_10400000. Forwarding
THREAD|17317.407|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17323.426|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17329.446|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000,
THREAD|17332.378|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17332.413|7 CRIT [CRIT]-MAC-----: NetworkTime: PrepareDataFrame offset=00000000_00000000, GetTimeSyncSeq=0
THREAD|17332.413|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17332.426|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17335.478|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17337.384|15 CRIT [CRIT]-CORE----: NetworkTime: TimeSync::GetTime: offset:20010C3C_10400000. Forwarding
THREAD|17367.379|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17368.373|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17369.373|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17370.373|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17370.765|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17371.515|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17372.765|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17375.309|15 CRIT [CRIT]-CORE----: NetworkTime: Newer time sync seq:5 received, offset:20010C3C_10400000. Forwarding
THREAD|17375.309|15 CRIT [CRIT]-CORE----: NetworkTime: TimeSync::GetTime: offset:20010C3C_10400000. Forwarding
THREAD|17378.792|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17384.809|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17390.831|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17392.378|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17392.400|7 CRIT [CRIT]-MAC-----: NetworkTime: PrepareDataFrame offset=00000000_00000000, GetTimeSyncSeq=0
THREAD|17392.400|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17392.412|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17393.792|7 CRIT [CRIT]-MAC-----: NetworkTime: clone offset=00000000_00000000, GetTimeSyncSeq=0
THREAD|17393.794|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17393.814|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17394.485|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17394.766|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=0 ***
THREAD|17394.792|7 CRIT [CRIT]-MAC-----: NetworkTime: SendTimeSync offset=00000000_00000000, GetTimeSyncSeq=0
THREAD|17394.794|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010C3C_10400000, timeIeOffset=21
THREAD|17395.264|15 CRIT [CRIT]-CORE----: NetworkTime: Newer time sync seq:7 received, offset:20010B0F_CECACBA9. Forwarding
THREAD|17395.264|15 CRIT [CRIT]-CORE----: NetworkTime: TimeSync::GetTime: offset:20010B0F_CECACBA9. Forwarding
THREAD|17395.792|7 CRIT [CRIT]-MAC-----: NetworkTime: SendTimeSync offset=00000000_00000000, GetTimeSyncSeq=0
THREAD|17395.794|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010B0F_CECACBA
THREAD|17396.136|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010B0F_CECACBA9, timeIeOffset=0 ***
THREAD|17396.161|7 CRIT [CRIT]-MAC-----: NetworkTime: send offset=20010B0F_CECACBA9, timeIeOffset=0 ***
```
If any additional logs are required, let me know, and I`ll try to reproduce this situation taking it to account.
 
Thanks!

librasungirl, 2020-01-13:
@AlexeyShako is the log customized in your build? I fails to find similar log info in existing OT repo
Note that not every frame is a time sync frame and only non-zero `timeIeOffset` indicates it is a frame for time sync service.

jwhui, 2020-03-09:
Closing stale issue.

closed 2020-03-09 by jwhui