openthread/openthread #4434: [coap] Retransmissions when talking to SED

dismirlian, 2019-12-30:
**Describe the bug**
When sending CoAP requests to a SED, the transaction has retransmissions. 
[image]

If the SED is turned into a REED or to a Rx-on-when-idle device, then the transactions work OK.
[image]

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id: 5b0af03afb8e70e8216f69623bd18bcd3d4b8b43
2. IEEE 802.15.4 hardware platform: efr32mg12
3. Build steps: make -f examples/Makefile-efr32mg12 BOARD=BRD4170A FULL_LOGS=1 COAP=1 LINK_RAW=1
4. Network topology: 1 FTD (leader), 1 FTD (configured as SED or REED), less than 1m apart.

On the Leader:
```
masterkey 00112233445566778899aabbccddeeff
panid 0xd7e0
mode rsdn
channel 26
ifconfig up
thread start
coap start
coap put <child's IP> a con 1234
```

On the child:
```
masterkey 00112233445566778899aabbccddeeff
panid 0xd7e0
mode s
channel 26
pollperiod 5000
ifconfig up
thread start
coap start
coap resource a
```

**Expected behavior**
CoAP transactions without retransmissions.

**Console/log output**
Please see the [Wireshark capture](https://github.com/openthread/openthread/files/4010254/coap-capture.pcapng.zip). The child was initially  changed to a Rx-on-when-idle MTD, then back to a SED, then to a FTD. Only when the device is a SED, the retransmissions occur.

**Additional context**
Add any other context about the problem here.

jwhui, 2020-01-10:
The retransmissions are due to CoAP retransmissions. Sleepy End Devices (SEDs) receive messages from their parent by sending an IEEE 802.15.4 Data Request message. The period between Data Request transmissions determines the latency that messages can be delivered to a SED.

In this case, it seems the CoAP ack timeout is shorter than the period that the SED sends Data Request messages. You can find the CoAP protocol parameter configuration in [src/core/config/coap.h](https://github.com/openthread/openthread/blob/master/src/core/config/coap.h). Currently, the configuration is compile time and affects all users of CoAP. I believe @sjlongland was looking into making those parameters configurable at run-time on a per-request basis.

dismirlian, 2020-01-10:
Hi @jwhui, thanks for your answer, it's clear now. For anyone who may encounter the same behavior, I leave here some additional info:

- The sender times out before the first data poll because the ACK timeout is 2~3s with the default values, so it queues the retransmission to the SED.
- The SED polls for data, receives the two PUTs (first transmission and retransmission) and responds.
- The sender receives the responses. The second retry had not yet timed out (4~6s), so the transaction completes.

> I believe @sjlongland was looking into making those parameters configurable at run-time on a per-request basis.

I second @sjlongland on this, it would be a very useful feature. @sjlongland, have you made any progress on this? Do you want me to look into it?

sjlongland, 2020-01-10:
On 11/1/20 5:22 am, Diego Ismirlian wrote:
>> I believe @sjlongland was looking into making those parameters configurable at run-time on a per-request basis.
> I second @sjlongland on this, it would be a very useful feature. @sjlongland, have you made any progress on this? Do you want me to look into it?

It's on the back-log, somewhere… I've been up to my eyeballs with a few
other things lately (particularly CoAP Observe) but I expect I'll be
getting onto that soon as we'll need it for a RS-485 serial to Ethernet
feature we're building.

(We actually need to speed things up, not slow them down like for SEDs,
but same problem, it needs to be done per-request.)

So I expect it'll be in the next few weeks.  You're most welcome to have
a look yourself. :-)

dismirlian, 2020-01-10:
Ok great, I'll close this and open a new issue so that we can share ideas. Thanks!

closed 2020-01-10 by dismirlian