openthread/openthread #4442: Move API group definitions out of error.h to a new file

greg-fer, 2020-01-02:
**Issue description**
The OT API documentation group hierarchy is currently defined in error.h file (lines 44-172):
https://github.com/openthread/openthread/blob/master/include/openthread/error.h

This is not the immediate and most obvious choice when working with API documentation and makes adding new groups more complicated.

**Expected behavior**
The ideal solution would be to have the API structure defined in a separate .h file, for example _api_structure.h_ or _ot_api.h_

LuDuda, 2020-01-02:
FYI @Vyrastas

Vyrastas, 2020-01-02:
I'll leave it up to @jwhui where it should go... would documentation.h make more sense?

jwhui, 2020-01-08:
I think moving the doxygen-specific group hierarchy out of `include/openthread/error.h` makes sense. Given that this is doxygen-specific, would it make sense to have such information located in [`doc/`](https://github.com/openthread/openthread/tree/master/doc) somewhere?

Vyrastas, 2020-01-09:
Yes, let's put it there.  I can move it.  Any preference on filename?

greg-fer, 2020-01-09:
Yes, having it in `doc/` makes sense. I suggest `api_doc.h` as the filename.

jwhui, 2020-01-10:
`doc/ot_api_doc.h` sounds good to me!

jwhui, 2020-01-10:
@greg-fer @Vyrastas, one of you want to take this on?

greg-fer, 2020-01-10:
@jwhui , @Vyrastas , feel free to take this on. I'm not that familiar with the OT project, and won't be able to work on this in the upcoming days/weeks. However, please add me to the PR if this is OK.

Vyrastas, 2020-01-14:
@jwhui So doing this messes with Doxygen a little because I run it off the /output/include/openthread directory of the Posix build.  Moving it to /doc means it's dropped in the build output.  What's the best way to get it included?  (Or should we just leave it as a separate file in the /include directory?)

jwhui, 2020-01-14:
@Vyrastas, how about adding the following:
```
openthreaddir = $(includedir)/openthread
dist_openthread_HEADERS = ot_api_doc.h
```
somewhere after:
https://github.com/openthread/openthread/blob/05ba4c29f4badfe99fe6b478900e7af072059ebe/doc/Makefile.am#L78

greg-fer, 2020-01-15:
If you agree, let's keep it in the `include/` directory.  This would make it easier to import it with other files in the `include/` directory to different projects.

Vyrastas, 2020-01-15:
@jwhui Your suggestion didn't work.

It might be easier to just leave it in `include/` but in a separate file.

jwhui, 2020-01-15:
@Vyrastas, is `--enable-docs` included as part of the build? That is required for `OPENTHREAD_BUILD_DOCS` to be true.

Otherwise, I am fine to leave it in `include/openthread` as a separate file.

Vyrastas, 2020-01-15:
Thanks, figured it out

PR #4476 (2020-02-03): [docs] move Doxygen group structure to the docs directory
Fixes #4442
closed 2020-02-03 by jwhui