openthread/openthread #4449: MTD configured NCP cannot start joining procedure

konradderda, 2020-01-04:
**Describe the bug**
When the NCP device IS NOT the FTD it fails to join a network. When the "join" command is used wpantund sets the "router eligible" attribute to false but it cannot be done due to "IsFullThreadDevice() check" in the "MleRouter::SetRouterEligible" method.

**To Reproduce**
Information to reproduce the behavior, including:
- Latest OT revision (5b0af03afb8e70e8216f69623bd18bcd3d4b8b43)
- Tested on nRF52840 but the bug is not platform-related
- Standard NCP build without any options
- Single device required

1. Start the NCP without any network 
2. Set "Thread:DeviceMode" to 0xD (if FTD)
3. Use the "join" command from wpanctl

**Actual behavior**
The NCP is unable to start the joining process. It fails at the beginning. NCP return the OT_ERROR_NOT_CAPABLE error code.

**Expected behavior**
The device starts the joining process.

**Console/log output**
Joining WPAN "TestNetwork" as node type "end-device", channel:16, panid:0xFACE
join failed with error 15351835. NCP-Specific Errorcode

konradderda, 2020-01-04:
I do not see any problem with setting router-eligible attribute from false to false for MTD devices since it does not impact anything. 

Proposed fix:  https://github.com/openthread/openthread/pull/4450

jwhui, 2020-01-06:
@konradderda, thanks for raising this issue and submitting a fix!

closed 2020-01-06 by jwhui