openthread/openthread #4459: MAC counters doesn't count out of bound tx frames

bukepo, 2020-01-09:
**Is your feature request related to a problem? Please describe.**
tx out of bound frames are also 15.4 frames, no matter how it is secured, the frame type can be retrieved. Shall we count these frames?

**Describe the solution you'd like**
count frames according to the frame type instead of tx operation.

**Describe alternatives you've considered**
count these frames as other frames because there's other rx frame counter.

**Additional context**
This is helpful to analyze power consumption.

bukepo, 2020-01-09:
@abtink @jwhui do you think out of bound frames should also be counted?

abtink, 2020-01-10:
Sounds reasonable to me.

jwhui, 2020-01-10:
Can you define "out-of-bound"? Do you mean "out-of-band"? If the latter, I agree that we should account for them.

bukepo, 2020-01-10:
sorry for the typo. Yes, it's out of band.

FiveDimensions, 2020-05-07:
add tx oob counter in PR #4936

librasungirl, 2020-05-17:
> Is your feature request related to a problem? Please describe.
> tx out of band frames are also 15.4 frames, no matter how it is secured, the frame type can be retrieved. Shall we count these frames?
> 

> Describe the solution you'd like
> count frames according to the frame type instead of tx operation.

The request requires counting OoB frame.
A bit ambiguity on the suggested solution - is it to count Oob by frame type? or redesign the mac counter to count all mac frame by frame type?
@bukepo

bukepo, 2020-05-17:
My original proposal was count by frame type, i.e. data frame, mac data request frame, etc.
@abtink Did we agree on this?

abtink, 2020-05-17:
@bukepo , either way (counting all as `mTxOther` or parse and consider per type) sounds ok to me
My suggestion https://github.com/openthread/openthread/pull/4936#discussion_r423917442 is mainly to keep the code simple (since we may remove the whole OOB feature later on).

bukepo, 2020-05-19:
I also OK to count as mTxOther. Thanks

PR #4936 (2020-05-21): [mac] count out-of-band Tx frames
Fix #4459 : MAC counters doesn't count out of band tx frames
closed 2020-05-21 by jwhui