openthread/openthread #4461: Use fixed endian for persisting settings?

bukepo, 2020-01-10:
**Is your feature request related to a problem? Please describe.**
Currently settings are persisted as opaque bytes, which is making restoring dataset from NCP not easy. I think this may also cause problem when switching from a 32-bit mode to a 64-bit mode. The alignment may be different.

**Describe the solution you'd like**
Implement a serializer and deserializer for settings structures, so that the data is independent of CPU architecture.

**Describe alternatives you've considered**
Allow platform to provide serializer and deserializer.

**Additional context**
Some products may migrate from NCP to RCP solution, some may switch from 32bit to 64bit.

bukepo, 2020-01-10:
@abtink @jwhui @librasungirl Thoughts?

abtink, 2020-01-10:
I like this. 
We should be careful of transition but would be good to have this for longer term.

jwhui, 2020-01-10:
Can you describe the challenge of restoring dataset from NCP?

My primary concern is the added overhead of needing to serialize the information. On embedded platforms, I don't think switching between 32 and 64 bit is a real use case.

bukepo, 2020-01-14:
The current status:

* The data structure of settings are **not packed**
* All settings use the CPU endianness.

As a result, the settings data is not portable. Besides, since the data structure is not packed, there may be paddings between fields and **after the end of data structure**.

For example `sizeof(ChildInfo)` is not 15, but 16(there's 1 byte padding). If a `uint8_t mSomeThing` field is added, `sizeof(ChildInfo)` is still 16, so the `mSomeThing` is uninitialized or meaningless. This is true on **single SoC platforms** as well. We have to ensure every time we add a new field, the structure `sizeof` value is increased, so that we can detect the change at runtime.

For the concern about burden. I didn't mean adding a heavy serializer to all platforms, but only want to make settings data portable.

1. Use packed data structures and fixed endianness. This should work for embedded platforms.
2. Allow platforms to provide customized serializer(compile time). This gives RCP/Host solution more choices, such as some human readable formats.

Thoughts?

jwhui, 2020-01-14:
> The current status:
> 
> * The data structure of settings are **not packed**
> * All settings use the CPU endianness.
> 
> As a result, the settings data is not portable. Besides, since the data structure is not packed, there may be paddings between fields and **after the end of data structure**.

Data that remains local to the same SoC generally does not need to be concerned about processor architecture differences.

> For example `sizeof(ChildInfo)` is not 15, but 16(there's 1 byte padding). If a `uint8_t mSomeThing` field is added, `sizeof(ChildInfo)` is still 16, so the `mSomeThing` is uninitialized or [...]

The structure is initialized via `memset` on `sizeof(ChildInfo)`, so `mSomeThing` would be initialized to zero.

https://github.com/openthread/openthread/blob/51485d4ecd0b008518538c9a5c6d7970c2ee1745/src/core/common/settings.hpp#L130

> For the concern about burden. I didn't mean adding a heavy serializer to all platforms, but only want to make settings data portable.
>
> 1. Use packed data structures and fixed endianness. This should [...]

What is the overhead of Option 1? It's difficult to evaluate a cost-benefit tradeoff without knowing the cost.

PR #4527 (2020-02-08): [settings] encode as packed little endian structs
The choice of little-endian was made for the following reasons:

- The existing encoding remains unchanged for settings unchanged on little-endian platforms.

- Eliminates any code overhead for little-endian platforms.

Resolves #4461
closed 2020-02-08 by jwhui