openthread/openthread #4462: [EFR32] Automake and autoconfig do create $(SDKVERSION) directory

NickNamerr, 2020-01-10:
**Describe the bug**
make examples/Makefile-efr32mg12 creates subdirectory
openthread/build/efr32mg12/third_party/silabs/gecko_sdk_suite/$(SDKVERSION)/

**To Reproduce**
Information to reproduce the behavior, including:
1. 73e30d1d7b1454e0d82607381b6b9c1f79ff4cec
2. EFR32MG12P432F1024GM48
3. ./bootstrap; make -f examples/Makefile-efr32mg12 BOARD=BRD4304A
4. NA

**Expected behavior**
Cannot complete building process.

**Console/log output**
`  BUILD    efr32mg12
make -j10 -C build/efr32mg12 --no-print-directory all
  CHECK         .local-version
make --no-print-directory all-recursive
Making all in include
make  all-recursive
Making all in openthread
Making all in platform
make[6]: Nothing to be done for 'all'.
make[6]: Nothing to be done for 'all-am'.
Making all in third_party
Making all in silabs
Makefile:970: gecko_sdk_suite/v2.7/hardware/kit/common/bsp/.deps/libsilabs_efr32mg12_sdk_a-bsp_bcc.Po: No such file or directory
Makefile:971: gecko_sdk_suite/v2.7/hardware/kit/common/bsp/.deps/libsilabs_efr32mg12_sdk_a-bsp_init.Po: No such file or directory
Makefile:972: gecko_sdk_suite/v2.7/hardware/kit/common/bsp/.deps/libsilabs_efr32mg12_sdk_a-bsp_stk.Po: No such file or directory
Makefile:973: gecko_sdk_suite/v2.7/hardware/kit/common/bsp/.deps/libsilabs_efr32mg12_sdk_a-bsp_stk_leds.Po: No such file or directory
Makefile:974: gecko_sdk_suite/v2.7/hardware/kit/common/bsp/.deps/libsilabs_efr32mg21_sdk_a-bsp_bcc.Po: No such file or directory
Makefile:975: gecko_sdk_suite/v2.7/hardware/kit/common/bsp/.deps/libsilabs_efr32mg21_sdk_a-bsp_init.Po: No such file or directory
Makefile:976: gecko_sdk_suite/v2.7/hardware/kit/common/bsp/.deps/libsilabs_efr32mg21_sdk_a-bsp_stk.Po: No such file or directory
Makefile:977: gecko_sdk_suite/v2.7/hardware/kit/common/bsp/.deps/libsilabs_efr32mg21_sdk_a-bsp_stk_leds.Po: No such file or directory
Makefile:978: gecko_sdk_suite/v2.7/platform/Device/SiliconLabs/EFR32MG12P/Source/.deps/libsilabs_efr32mg12_sdk_a-system_efr32mg12p.Po: No such file or directory
Makefile:979: gecko_sdk_suite/v2.7/platform/Device/SiliconLabs/EFR32MG12P/Source/GCC/.deps/libsilabs_efr32mg12_sdk_a-startup_efr32mg12p.Po: No such file or directory
Makefile:980: gecko_sdk_suite/v2.7/platform/Device/SiliconLabs/EFR32MG21/Source/.deps/libsilabs_efr32mg21_sdk_a-system_efr32mg21.Po: No such file or directory
Makefile:981: gecko_sdk_suite/v2.7/platform/Device/SiliconLabs/EFR32MG21/Source/GCC/.deps/libsilabs_efr32mg21_sdk_a-startup_efr32mg21.Po: No such file or directory
Makefile:982: gecko_sdk_suite/v2.7/platform/emdrv/dmadrv/src/.deps/libsilabs_efr32mg12_sdk_a-dmadrv.Po: No such file or directory
Makefile:983: gecko_sdk_suite/v2.7/platform/emdrv/dmadrv/src/.deps/libsilabs_efr32mg21_sdk_a-dmadrv.Po: No such file or directory
Makefile:984: gecko_sdk_suite/v2.7/platform/emdrv/gpiointerrupt/src/.deps/libsilabs_efr32mg12_sdk_a-gpiointerrupt.Po: No such file or directory
Makefile:985: gecko_sdk_suite/v2.7/platform/emdrv/gpiointerrupt/src/.deps/libsilabs_efr32mg21_sdk_a-gpiointerrupt.Po: No such file or directory
Makefile:986: gecko_sdk_suite/v2.7/platform/emdrv/uartdrv/src/.deps/libsilabs_efr32mg12_sdk_a-uartdrv.Po: No such file or directory
Makefile:987: gecko_sdk_suite/v2.7/platform/emdrv/uartdrv/src/.deps/libsilabs_efr32mg21_sdk_a-uartdrv.Po: No such file or directory
Makefile:988: gecko_sdk_suite/v2.7/platform/emdrv/ustimer/src/.deps/libsilabs_efr32mg12_sdk_a-ustimer.Po: No such file or directory
`

**Additional context**
If manually rename "openthread/build/efr32mg12/third_party/silabs/gecko_sdk_suite/$(SDKVERSION)/" to "openthread/build/efr32mg12/third_party/silabs/gecko_sdk_suite/v2.7/" - building process is going with no issue.

marvensilabs, 2020-01-10:
Hi Nick,

We have seen this issue overnight on our internal build system and are investigating this.  It has been seen that it creates a $(SDKVERSION) directory within the build directory on the build system but on my system this is replaced with v2.7.

I prefixed with "export" as without it on my system it appeared not to be passed down correctly to some of the sub-makefiles, in particular the third_party/silabs where /v2.7/ was expected in pathnames I had build errors with it appearing as //.

marvensilabs, 2020-01-10:
Further testing on macOS and Linux is showing it building with Automake 1.16.1 whereas one of failing builds is using Automake 1.15.1.   If I use the -k option on make command line it skips the errors where it builds the dependency files and continues to build the sources and final images.  However it fails to copy the built files to the output directory.

It is as if Automake 1.15.1 does not expand $(SDKVERSION) in the  _SOURCES files in third_party/silabs/Makefile.am when creating dependencies.

marvensilabs, 2020-01-10:
One option is to replace the gecko_sdk_suite symlink in third_party/silabs with a symlink to the v2.7 directory within gecko_sdk_suite instead and remove the SDKVERSION variable.

marvensilabs, 2020-01-10:
I intend to remove SDKVERSION for the moment and replace it's use with v2.7.   At later date this can be looked at further as we consider how to integrate it into the Silicon Labs SDK.

jwhui, 2020-01-13:
Resolved by #4466

closed 2020-01-13 by jwhui