openthread/openthread #4470: [coap

dismirlian, 2020-01-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

sjlongland, 2020-01-11:
My thinking is, you're not necessarily going to have every message with different timing parameters, in fact I think the vast majority of users of the CoAP library will be using the defaults (from RFC7252).

I think it fair to assume that the application developer would define a few parameters as a `const struct` and pass that in somehow.  Thus we could have a "default" one exported by OpenThread, and if someone wants something different, they declare it and tell the CoAP library what they need for that request.  My thoughts were either an extra parameter to `otCoapSendRequest`, which could be `NULL` to accept defaults.

There's a few derived constants, two approaches could be used here:
- define variables to hold the computed values of these derived constants (chews up maybe a dozen bytes or two per request "slot")
- compute the values on demand (a few more processor cycles consumed… the calculations are not particularly strenuous)

dismirlian, 2020-01-11:
> My thinking is, you're not necessarily going to have every message with different timing parameters, in fact I think the vast majority of users of the CoAP library will be using the defaults (from [...]
Exactly my thoughts

> There's a few derived constants, two approaches could be used here:
>
> * define variables to hold the computed values of these derived constants (chews up maybe a dozen bytes or two per request [...]
Browsing the code, it seems that there are very few calculations to be done, and (maybe almost) all during creation of the new message. I'm thinking that with a few minor modifications, the CoapMetadata class can hold the required status in the existing member variables:
- mRetransmissionTimeout (can be calculated on "transaction" creation from ACK_TIMEOUT and ACK_RANDOM_FACTOR)
- mRetransmissionCount (we could change this to mRemainingRetransmissions to avoid also saving MAX_RETRANSMIT, and compare with 0).

jwhui, 2020-01-13:
@dismirlian @sjlongland, thank you for looking into this enhancement. The proposed path forward seems reasonable to me.

jwhui, 2020-02-22:
Resolved by #4481

closed 2020-02-22 by jwhui