openthread/openthread #4471: [diags]

zhanglongxia, 2020-01-13:
#### Is your feature request related to a problem? Please describe.

The format and content of the frames sent by `factory_diag` are fixed. When we are debugging the radio driver, we need to generate real 15.4 frames which include FCF, Seq and MAC header to verify whether the radio driver could successfully handle different kinds of frames.

#### Describe the solution you'd like
Add more commands to `factory_diag` to set the parameters of the 15.4 frame. The `factory_diag` generates sent frames based on the set parameters. 
The new added commands: 
"panid"  : set PanId
"srcaddr" : set MAC source address
"dstaddr" : set MAC destination address
"csmaca" : enable CSMA/CA
"ackreq" :  set the ACK request bit filed
"promiscuous" : set promiscuous mode
"format" :  indicate whether generate the 15.4 frame based on the set parameters

#### Describe alternatives you've considered
None.

zhanglongxia, 2020-01-13:
@jwhui @abtink  @bukepo  Thoughts?

jwhui, 2023-06-27:
Closing stale issue.

closed 2023-06-27 by jwhui