openthread/openthread #4472: Packets sent to Service ALOC not reaching destination

mwilniewiec, 2020-01-13:
Lately an issue with Service ALOCs was found and fixed #4424 

Currently we observe second part of the same problem. 
A child service registers its RLOC in the Leader and it is added to the list but all old RLOCs are not removed. As a result we have 1 server node but multiple Server TLV entries in the network data mostly pointing to some old RLOCs.

During ALOC routing some routers route packet to a valid RLOC but some pick an invalid one and packets are lost.

Im missing something or specification is not addressing this problem too.

librasungirl, 2020-01-13:
@mwilniewiec would you please describe the process a bit more when the issue happens? e.g. Did this child server attach to a different parent?

mwilniewiec, 2020-01-13:
Yes the issue arrises when mesh topology is changing. Our server becomes a child of a different parent or changes chlid->router or router->child (generally changes its RLOC). 
When mesh boots up for the first time - server registers itself and leader has only 1 RLOC in network data. When server changes its RLOC it notifies Leader about the change and a new Server TLV is added to the network data.
We ended up with 1 server having 8 different RLOCs is network data. Some garbage collection is needed as there is no chance to route a packet properly in such situation.

librasungirl, 2020-01-13:
@mwilniewiec Would you please share what the 8 different RLOCs look like?

RLOC is considered temporary in the Thread Mesh Network. It may take some time for the outdated network data info to be removed from the Leader Network Data.

1) If child attaches to a different router, the old parent would deregister the outdated service when timing out the child (up to the ChildTimeout reported during attach, 240s by default).
    [To remove the depracated service sooner, proper shorter Child Timeout might help]

2) If child->router or router->child, when it registers its service, it would include its old RLOC16 when updating its service (https://github.com/openthread/openthread/blob/master/src/core/thread/network_data.cpp#L998) , thus the old server entry would be removed by the Leader before new entry is added.

mwilniewiec, 2020-01-13:
@librasungirl Sure, this is the network data I've taken from one node:
`0b23
000000c6ac0100
0d 02 b809
0d 02 c812
0d 02 9411
0d 02 f81a
0d 02 ac0c
0d 02 8c0c
0d 02 8c17
080b
0b02 b852
0801 ff
0902 8c17`
You can notice 7 RLOCs here here only one is valid:
0d 02 c812
0d 02 9411
0d 02 f81a
0d 02 ac0c
0d 02 8c0c
0d 02 8c17

Our setup has only one server (NCP) with 30 other nodes. Issue is persistent for at least several hours. Invalid RLOCs aren't disappearing after 240s for sure. Im reproducing the same issue on 10 nodes setup too.

librasungirl, 2020-01-14:
@mwilniewiec 
All the Server RLOC16 in the network data are Child's RLOC16, it seems somehow the parents didnot de-register the service on behalf of its child.

Would you check whether or not the child table of old parents (e.g. Router c800) still has valid entry for the server? 

Just fyi,  [one test script](https://github.com/openthread/openthread/compare/master...librasungirl:test_child_service4issue4472) is writtten to simulate the scenarios of attach to a different parent or upgrade to router or downgrade to child . Here the ([pcap](https://drive.google.com/open?id=1_TUkmoK_t9Za0BiUmdmYHAf8KkYPCwcs)) validates the two points I mentioned above and there is only one server entry in the Service TLV.

mwilniewiec, 2020-01-14:
@librasungirl 
I managed to reproduce the issue on master, posix 4x ot-ftd-cli with similar setup to yours (a leader, 2x router and a child server).

```
PARENT1 --- LEADER --- PARENT2
       \             /
         \         /
        SERVER (child)
```

The clue is that some packet must be lost which is normal on hardware nodes and not normal on posix. 
I reproduced persistent multiple RLOCs in network data only after setting some blacklists between routers and leader just before repairing packets are sent. I disabled them later not to lose those routers from the router table.

Currently Im not sure the exact reproduction path as I tried to do the same 1 more time and I failed.
Tomorrow I'll try to find step by step reproduction. Still you could now have a new point of view for that issue.

librasungirl, 2020-01-15:
@mwilniewiec Do you have sniffer in your setup? The capture may help for analysis.

librasungirl, 2020-01-17:
Thanks for @simonlingoogle for reproducing the issue by simulating certain packet loss (30%) with the OTNS #4268 
@mwilniewiec I have had the pcap now and confirmed the issue exists. 

Previous I thought 1) there are two chances to deregister the invalid service of old child - one is the old parent deregister it when timing out the child, the other is the child itself will include old rloc16 when register  its service to leader after reattach. And 2) CoAP retransmissions could help the delivery of service registration in network with packet loss. However the traffic shows that the CoAP cache and retransmission may cause issues when rloc changes in high packet loss network.

Still need some time to figure out the solutions to get the Thread Network Data corrected.

mwilniewiec, 2020-01-17:
Im glad to hear that my investigation helped. My thoughts are:

1. All nodes could track RLOCS in network data that belongs to them and notify leader: There is my RLOC in network data but Im not the server.
2. Routers could track RLOCS in network data and notify leader: There is an invalid RLOC as this child doesn't exist

In that case no special retransmission is needed and its just some suggestion :)
Please let me know on any progress on that issue and it is quite blocking for us and we need to find a fix quickly or change our implementation from services to something else.

librasungirl, 2020-01-17:
@mwilniewiec
Just submitted https://github.com/openthread/openthread/pull/4485 to relieve the stale network data issue in high packet loss network. 

Would you please help to check if it helps for your setup?

Nopik, 2020-01-18:
Looks good to me.

mwilniewiec, 2020-01-21:
@librasungirl hi again, 
We applied your patch to our hardware setup and we can still reproduce the issue:
For more than an hour now we have 2 RLOCs in network data (with only 1 server):
Full network data on a node in the network:
`0b0f000000c6ac01000d02f4070d02540808040b022206`
where we have 2 RLOCs:
0d 02 f407 (some invalid value, not replying to a ping)
0d 02 5408 (valid server RLOC, replying to a ping)

What is more we tested that on a 540A node which is a child of the same router as 5408 and  FC10 ALOC packets didn't reach 5408. I created a separate issue for that #4496

mwilniewiec, 2020-02-06:
@librasungirl Im back :)
We applied all your current patches for this issue and after longer testing we reproduced the issue.
It looks like leader is not clearing its own network data properly when there is an invalid RLOC in services (child of non existing router in our case). Our leader was its own leader with no routers and no neighbors but it's network data had some old RLOC. Before we removed antenna from our leader  it had a mesh of 8 nodes and all of them received network data with Service RLOC = e006 and there was no e000 router in the router table.

Below you can see that a leader with no neighbours has 0d 02 e006 in his network data.

`> state
leader
Done
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
| 62 | 0xf800 |       63 |         0 |     0 |      0 |   0 | e6c480a2743d6d54 |
Done
> neighbor table
| Role | RLOC16 | Age | Avg RSSI | Last RSSI |R|S|D|N| Extended MAC     |
+------+--------+-----+----------+-----------+-+-+-+-+------------------+
Done
> networkname
biotcloud-4
Done
> scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
> | 1 | biotcloud-30     | d6f778b09b829ec5 | 4c60 | eeca2dee7d3a4bff | 13 | -95 |   4 |
| 1 | biotcloud-30     | d6f778b09b829ec5 | 4c60 | 0e1e0a0dfd937cd1 | 13 | -95 |   4 |
| 0 | biotcloud-28     | a428c376121047c4 | b3fc | 164c20eea729bb07 | 15 | -90 |   8 |
| 0 | biotcloud-28     | a428c376121047c4 | b3fc | fa3a16adc3e807a3 | 15 | -88 |  20 |
| 0 | biotcloud-28     | a428c376121047c4 | b3fc | 3e6ed181ccc2d872 | 15 | -77 |  64 |
| 0 | biotcloud-28     | a428c376121047c4 | b3fc | 06d81e16425e8910 | 15 | -79 |  60 |
| 0 | biotcloud-28     | a428c376121047c4 | b3fc | e666c5f703159f5f | 15 | -79 |  56 |
| 0 | biotcloud-28     | a428c376121047c4 | b3fc | 321be7901a520a09 | 15 | -85 |  32 |
| 0 | biotcloud-11     | 127d4d3018867191 | 4fa4 | 6ed129fdbdf3a5a7 | 20 | -95 |   4 |
| 0 | biotcloud-30     | ee2224ba3bd43617 | cf6b | 3e3b8cb0cf7ad74c | 20 | -92 |   8 |
| 0 | biotcloud-30     | ee2224ba3bd43617 | cf6b | b6e515d457a07936 | 20 | -89 |  16 |
| 0 | OpenThreadBiot1  | 1aaa456789abcde1 | abc1 | 8284b69ee69d0cff | 21 | -96 |   4 |
| 1 | biotcloud-31     | 789c1b8c810e45c8 | 7fb3 | 5232a28fb4e3c2d7 | 21 | -93 |   4 |
| 1 | biotcloud-6      | 3e0385b96cc047fc | e121 | 96d835a565b8bdb4 | 23 | -93 |   4 |
| 1 | biotcloud-6      | 3e0385b96cc047fc | e121 | 76198f9a6eaeee70 | 23 | -93 |   4 |
Done
> netdatashow
0b0b000000c6ac01000d02e00608040b02a10d
Done
> `

If you need I have a pcap but only from sniffing the orphan leader not the whole process of loosing e000 etc.

librasungirl, 2020-02-08:
@mwilniewiec , thanks for the updates.

For Child Server of non-existing router, in theory, the Leader would remove any network data relative to this non-existing router and its child when removing this router in existing implementation:
https://github.com/openthread/openthread/blob/master/src/core/thread/router_table.cpp#L295

Not sure how it happens. But please share the pcap if you have, let's see if it helps.
One more question is - Does the invalid service stick there all the time?

mwilniewiec, 2020-02-10:
@librasungirl, thanks for the update
Code in the link which you sent looks OK. I also verified our openthread codebase as we use older version and there is such code too.
One thing that comes to my mind is that maybe router/child is sending a Service registry to the leader after its removed from router table. Would it be possible in a high packet loss setup?

1. Node registers itself as a  service in the leader (network data)
2. Connection is bad and the leader loses the router of the service
3. Leader removes a router from a router table and notifies about a network data change
4. Service node somehow receives a network data change and registers itself again

Its very hypothetical but it must have happened somehow.

jwhui, 2020-02-10:
> @librasungirl, thanks for the update
> Code in the link which you sent looks OK. I also verified our openthread codebase as we use older version and there is such code too.
> One thing that comes to my mind is that maybe router/child is sending a Service registry to the leader after its removed from router table. Would it be possible in a high packet loss setup?
> 
> 1. Node registers itself as a  service in the leader (network data)
> 2. Connection is bad and the leader loses the router of the service
> 3. Leader removes a router from a router table and notifies about a network data change
> 4. Service node somehow receives a network data change and registers itself again
> 
> Its very hypothetical but it must have happened somehow. PS. tomorrow I'll do some more testing of ALOCs in a bigger setup of 50 nodes. Will let you know about the results.

Good catch. I don't think we have a check that rejects registrations with an unallocated Router ID. Would you like to propose a patch? Otherwise, I'm happy to submit a proposed patch for this specific issue.

jwhui, 2020-02-10:
@mwilniewiec, I just submitted $4544 to address the issue of registering network data with invalid router id. Please review.

librasungirl, 2020-02-17:
@mwilniewiec 
https://github.com/openthread/openthread/pull/4485 is rebased on master branch with @jwhui 's fix, and it also newly introduce the network data detection for invalid child server when there is new network data.

If you are stilling work on the service solution recently, would you please help to try it out? Thanks.

mwilniewiec, 2020-02-18:
Applied all your and Jonathans patches to our devices and the issue wasn't reproduced for a few days now. Thanks for all the fixes, in case I see something related to the same problem I can open a new issue.

jwhui, 2020-02-24:
Resolved by #4485

closed 2020-02-24 by jwhui