openthread/openthread #4479: Child Supervision disable meaning

bhumikaajain, 2020-01-16:
In openthread stack, by default child supervision is disabled.
So how or when does mtd re-attach process is enabled in case child supervison is disabled.

abtink, 2020-01-16:
For Child Supervision I can refer you to https://openthread.io/guides/build/features/child-supervision.
Child supervision feature is not (directly) related to how re-attach process works. It is a mechanism for the child to detect whether the parent has removed it from its child table. 

Regarding re-attach process,  a detached (MTD) child would continuously try to search for a parent (router) to attach to. Feature `OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_ENABLE` enables  addition of an exponentially increasing backoff wait time between attach attempts.

bhumikaajain, 2020-01-16:
can you please explain to me difference in polling and child supervision?

abtink, 2020-01-16:
They are two different functionalities (not related):
- Polling (IEEE 802.15. data request) -> Sleepy child send data polls to its parent to query about any pending message for it (queued on the parent) and receive them.
- Child supervision ->  https://openthread.io/guides/build/features/child-supervision Feature/mechansim for child to detect if parent has timed out the child entry from parent's child table by ensuring parent sends at least one frame to the child within every supervision interval. 

Data polls are sent from child to parent. Supervision msg (when feature is enabled) are sent from parent to child

bhumikaajain, 2020-01-27:
I have used polling method in my stack and child supervision is disabled by default.
In this case, how will MTD know if parent has removed it from it's child table or not forwarding its data further.

jwhui, 2020-01-27:
The child does not know if the parent still has a child entry for it simply by sending data polls. The child needs to receive a data message from the parent to validate this (see https://github.com/openthread/openthread/issues/4478#issuecomment-575501278).

Also, the child does not know simply by sending messages to the parent. The only way to validate whether the parent is forwarding further is by using higher-layer acknowledgments, typically at the transport or application layers.

jwhui, 2020-02-18:
Closing stale issue.

closed 2020-02-18 by jwhui