openthread/openthread #4496: Invalid ALOC routing between router children

mwilniewiec, 2020-01-21:
Lately we are testing heavily service ALOCs on NRF52840 hardware.
We noticed a strange behaviour in a setup:

```
NCP (0x5408, FC10 service aloc)
              \
               \
                ROUTER (0x5400) --- SOME OTHER ROUTERS --- ....
               /
              /
NODE (0x540A)
```
In network data on 0x540A we can see 2 different RLOCS for a FC10 ALOC:
0d 02 5408 (which is ok)
0d 02 f407 (invalid RLOC as we only have 1 server in the network, see #4472 )

Still... communication to FC10 from 0x540A should work without any problems as 5400 should route FC10 packet to its closest child 5408.
We notice that these packets are not reaching NCP for more than an hour now and we suspect that there is something wrong with routing (there might be an error in cost calculation).

librasungirl, 2020-01-21:
@mwilniewiec do you have the pcap?

mwilniewiec, 2020-01-21:
Nope, we have no sniffer in our hardware setup. We must investigate how to make one first :)

librasungirl, 2020-01-21:
Agree that the parent router should find the closest child server. will look into this.
However not sure why the invalid server entry not being removed in such long time given #4485.

Fyi, https://openthread.io/guides/ncp/sniffer, any NCP device with LINK_RAW enabled or RCP device could act as the sniffer.

librasungirl, 2020-01-21:
@mwilniewiec If the issue could be easily reproduced, the pcap would surely help to accelerate the analysis process. thanks

mwilniewiec, 2020-01-22:
We have some suspicion that this cost calculations works not as it is expected in this part of code:
https://github.com/openthread/openthread/blob/c4f32fd9074708de6e95bfa75ddedcf4a1cee4ed/src/core/thread/mesh_forwarder_ftd.cpp#L797
If you could look into this, what is a calculated cost for own child or self. Maybe we some additional checks ("is it my packet?") are needed before calculating any costs.

librasungirl, 2020-01-23:
@mwilniewiec 
Just added a new commit to #4485  to resolve the child server selection issue, would you please have a try?

mwilniewiec, 2020-01-26:
Recently we used a workaround as we still had multiple service ALOCs in network data. When services are not used we observe more stable communication. Will test both fixes and provide pcap after we resolve other critical issues for us.

librasungirl, 2020-02-18:
[commit 6b3e45a](https://github.com/openthread/openthread/pull/4485/commits/6b3e45af71aa75d4df65360068a316b87bf4562a) in #4485 should have solved the original issue reported here about routing to the server of minimum cost, right? @mwilniewiec

mwilniewiec, 2020-02-18:
We did some testing with your patch on our 50 devices setup and couldn't see issues. We can consider it as fixed.

closed 2020-02-24 by jwhui