openthread/openthread #4508: Not stable Thread network topology when applying higher traffic

mwilniewiec, 2020-01-24:
Hi, first I must say it is very critical issue for us and I would need any urgent help if possible.

### The Setup:
We have a hardware setup of 50x nRF5280 + one NCP and a border router.
Currently we test them in a line topology:
`BR (NCP) --- NODE --- NODE --- NODE --- ... --- NODE`
Each link to a nearest neighbour is between 70 and 80 dbm, Nodes can see around 3-4 neighbour nodes (RSSI: 80-95).
All nodes send keep-alive confirmable coap to the NCP Mesh Local address every 1 minute.

### Observations:
Thread network topology becomes stable quite quickly (within single minutes),
Keep alives from all nodes reach border router which is fine.
Communication can be stable for hours with rare keep alives being lost.

When we push some messages into the network ( Mesh multicast + confirmable command to every node Mesh local address ) everything goes crazy.
Keep alives are all gone or almost all gone, we observed NCP parent router to change its RLOC (d000 ->bc00), looks like thread topology is changing/partitioning?.

Short higher traffic can break the network within seconds but it takes around 10 minutes to recover from that situation. After several minutes everything gets back to normal. 

### Notes
We didn't observe similar issue when topology wasn't linear (less hops).
We will provide more details ASAP like sniffer pcap, some test results, charts and anything that could help.

### Questions
Do you have any thoughts why this is happening?
Are there any timeouts we could tune to make it better?
Have you worked with similar hardware setups with many nodes and many hops?
@jwhui I would be greatful if you could look into the issue

Nopik, 2020-01-24:
This version uses `LONG_ROUTES` enabled (to allow 'deep' networks), as well as has `kAdvertiseIntervalMax ` set to `5`, as discussed in https://groups.google.com/d/msg/openthread-users/F6YY_lQiTZg/QOUAPa6LAAAJ

jwhui, 2020-01-24:
Can you further clarify "push some messages into the network"?
1. Are both the multicast and confirmable coap request initiated by the border router?
1. Are multicast and confirmable coap request messages generated by the border router to each individual device?
1. At what rate are multicast and confirmable coap request messages being generated?

mwilniewiec, 2020-01-24:
1. All the messages are sent from border router NCP.
2. Our testcase starts with one multicast to all nodes. 
There is a unicast retry to all nodes that didn't reply to multicast (confirmable coap)
3. Multicast start, unicast retries after 15 seconds, second retry to all with no answer after 20 sec, third 25, fourth 30 etc.
Problem generally occurs before second /third retry.
On monday I'll do some more tests with no multicasts (multiple unicasts only)

jwhui, 2020-01-25:
Thanks for the additional detail. Some additional questions:

1. Devices receiving the multicast message from the border router send a confirmable coap message back to the border router?
1. When the border router attempts unicast retries, does it generate a confirmable coap message to all devices that did not respond simultaneously (e.g. after 15 seconds, then again after 20 seconds, etc.)? Are these coap messages rate limited in any other way?

mwilniewiec, 2020-01-25:
1. Yes there is a confirmable coap node -> ncp when node receives a message (looks like a place to improvement, will test if non confirmable or piggyback answer causes the same issue)
2. In practice after 15 seconds router recives infomation from cloud to send messages to its nodes. Around 40 coap confirmable messages are sent almost immediately (not as fast a a forEach loop) but it's done on 40 separate messages received from cloud.

mwilniewiec, 2020-01-26:
It looks like without initial multicast this works better. Still managed to reproduce the issue 1 out of 5 with unicasts only but there is a noticable difference.
Firstly I thought it just a high traffic issue but we started a DFU upgrade on the same configuration causing thousands of messages sent and its more stable too.
What we also observed previously is that the same multicast can reach a node multiple times.
My suspition is that: 
1. A multicast is reaching the same node multiple times
2. Additionally it recives some unicast messages
3. On every received message it sends a confirmable coap to NCP Mesh local address
4. Its not receiving an answer (becase of connection issues?)
5. Confirmable messages queue becomes full
6. We start to have problems with this node

Does it make sense? Should it affect topology changes, mle messages?
Soon we will do some testing without this confirmable message.

jwhui, 2020-01-27:
IPv6 multicast messages should not be passed up multiple times - MPL uses an IPv6 Hop-by-Hop Option header with a sequence number to detect duplicate messages. One thing you could try is increasing [OPENTHREAD_CONFIG_MPL_SEED_SET_ENTRY_LIFETIME](https://github.com/openthread/openthread/blob/a5f2a41cb3f05d330269af6f86e4b7be05bc5c0e/src/core/config/ip6.h#L147), which specifies the number of seconds the (RLOC, sequence number) cache entries is held.

If you are **not** using RLOC addresses, another thing you could try is increasing [OPENTHREAD_CONFIG_TMF_ADDRESS_CACHE_ENTRIES](https://github.com/openthread/openthread/blob/a5f2a41cb3f05d330269af6f86e4b7be05bc5c0e/examples/platforms/nrf528xx/nrf52840/openthread-core-nrf52840-config.h#L122) to be at least as many nodes as you expect in your network. This should help avoid a unicast message from the NCP generating a multicast Address Query message being sent to discover the RLOC for a given IPv6 address.

The above ideas are just shots in the dark. Hopefully with more info, we can narrow into the root cause.

mwilniewiec, 2020-01-27:
First of all my theory failed ...
We see that packet isnt delivered multiple times to the node and using a non confirmable message node->ncp didn't help that much. I couldn't observe multicast being delivered multiple times in this setup too.

Still it seems that getting rid of the initial multicast has the biggest positive effect and mesh is more stable without it (some occasional topology changes).

We send all the messages ( 50 pcs ) in 100ms interval without an initial multicast. Our biggest problem now is that these messages are lost or their replies are lost and we are not seeing a reply on the cloud side. We must investigate first what is wrong with that.

Can you see any other magic consts for tuning that could help in a long route, lower rssi environment?

We changed OPENTHREAD_CONFIG_TMF_ADDRESS_CACHE_ENTRIES from 20 to 200 and will do some more tests and deeper investigation tomorrow.

jwhui, 2020-02-04:
@mwilniewiec, did you get a chance to test with a larger value for `OPENTHREAD_CONFIG_TMF_ADDRESS_CACHE_ENTRIES`? Any other observations you can share?

mwilniewiec, 2020-02-05:
Yes, we tried that its hard to say if bumping OPENTHREAD_CONFIG_TMF_ADDRESS_CACHE_ENTRIES from 20 to 200 did anything good but we left it that way.

We did some investigation with the sniffer and we found that our multicast, actually multicast answers were causing too big load on the network. Every node upon receiving multicast message was responding with multiple packets on NCP Mesh Local address. Additionally packets sent to Mesh Local were lost causing sending a/aq and a/an and causing even bigger mess in the network.
I believe traffic was so big that thread network couldn't handle that and it takes a few minutes to recover.

Currently we removed the multicast and use 50x unicasts to every node with 200ms delay for each and we try to reduce answers to the minimum. That helped and everything is more stable and reliable. We also plan to get back to sending messages to NCP ALOC instead of Mesh Local when we confirm that #4472 #4424 are fixed.

Still if its just a matter of heavy network load I believe there could be some improvements in openthread to favor MLE or other critical messages over application coap.

We are still investigating some issues and with such setup strange things happen from time to time. I will provide you with more info/issues when I'll be able to describe them properly.

jwhui, 2020-03-16:
@mwilniewiec, any new information you can share?

mwilniewiec, 2020-03-18:
I can share some interesting information not directly related with that issue:

Currently we have a test setup of 140 hardware nodes and a Border router with NCP.
Every node can see only a few neighbors (RSSI ~75dBm) so we have more linear physical topology.
When we find anything that can be described we create a separate issue for that, there were some already :)

Network is mostly stable. I still must sniff a bit more to verify how many topology changes we have.
Our application CoAP messages work fine as long as we use them carefully without flooding the network.

The number of 140+ seems to be a number where we start having problems with wpantund and NCP. Hangups, crashes, permanent death. This is under investigation now and some fixes were already done: buffer size tuning, #4641 .

I am quite surprised with the fact that we observe high packet loss (even on smaller setup like 5-10pcs). If we set them in a linear topology we lose lots of CoAP messages on the application layer even when those packets are very small and fit into 1 frame. Instead of receiving an answer immediately we often must wait for multiple retries.
This issue is most striking during thread joining process. There are multiple DTLS handshake messages which are quite big and with high packet loss its almost impossible for a node to join over multiple hops. This can take forever or at least hours of join tries.

I must do some more investigation to find out where the problem really lies. Is the signal too weak, are there too many collisions or is it something totally different.

Please share any of your thoughts.

Nopik, 2020-03-18:
Please also note, that this environment is using the `long routes` extension, allowing us to have 32 hops in the partition, up from standard 4.

mwilniewiec, 2020-04-06:
@jwhui I did a bit more investigation on network stability.
In such a 140 nodes network we have huge numbers in parent changes, better partition attempts. 
It  isn't helping at all when nodes change their RLOCs. 
Scale of problem:
- MLE Link, Parent, Child req/res/accept are even 33% of all sniffed packets while MLE Advertisement is 3%
- Nodes change their RLOC every few minutes (noticed while working with single nodes CLI etc)
- Nodes get into detached state from 1 to 5 times per hour (taken from thread counters)
- Better partition attempts occur up to 100 times per hour

Can you see anything that can be checked, changed to limit the problem somehow?

jwhui, 2020-04-07:
> I am quite surprised with the fact that we observe high packet loss (even on smaller setup like 5-10pcs). If we set them in a linear topology we lose lots of CoAP messages on the application layer [...]

Have you tried increasing `OPENTHREAD_CONFIG_MAC_MAX_FRAME_RETRIES_DIRECT` to a larger value? In Google Nest products, we set this to 15.

jwhui, 2020-04-07:
> In such a 140 nodes network we have huge numbers in parent changes, better partition attempts.
> It isn't helping at all when nodes change their RLOCs.
> Scale of problem:
> 
> * MLE Link, Parent, Child req/res/accept are even 33% of all sniffed packets while MLE Advertisement is 3% and Radio ACK is 30%
> * Nodes change their RLOC every few minutes (more personal feeling while working with single nodes CLI)
> * Nodes get into detached state from 1 to 5 times per hour (taken from thread counters)
> * Better partition attempts occur up to 100 times per hour
> 
> Can you see anything that can be checked, changed to limit the problem somehow?

The above seems to indicate that the network is not converging to a single partition. Some things to look for:

- **Device reboot**: normally the device should try to restore its previous state after a reboot. If the device cannot communicate with its neighbors, it will establish its own partition. Note that we recently fixed an issue that could address some poor performance here: #4753.

- **Failure to receive sequence number updates from Leader**: Thread routers determine connectivity to the leader by periodically receiving sequence number updates from the leader. If the Thread router is not receiving sequence number updates, it will detach and could form a new partition. Failure to receive updates generally due to lost MLE Advertisement messages. Sometimes the issue is more subtle. For example, in #2755, we had to decrease `kAdvertiseIntervalMax` to ensure that the information propagates with a more deterministic period, especially in large networks.

- **Failure to become a router after attaching as a end device**: a router-capable device must communicate a CoAP exchange with the Leader to obtain a valid router id. If the device is not able to communicate with the leader, any neighboring device trying to attach will not be able to and may form their own partitions.

Not sure if any of the above helps, just trying to brainstorm around what may be going on.

mwilniewiec, 2020-04-12:
@jwhui thanks for your ideas. I did proposed changes and will test the behavior soon. Our kAdvertiseIntervalMax is already 5 as we use Long routes in our setup.
From what I read there is a 120 timeout for receiving MLE over hops. What woud you say to increase it to 180 or 240 in our setup? I believe such changes could make the network more stable too.

mwilniewiec, 2020-04-18:
@jwhui I did some more investigation on devices which stay in a different partition unable to merge.
Im not sure yet what is the reason of initial disconnection but we have such case: 

1. Device sends a Parent Request for some reason (already in different partiton, lost his parent, lost a leader?)
2. bla bla
3. fdfdsfds

mwilniewiec, 2020-04-18:
@jwhui I did some more investigation on devices which stay in a different partition unable to merge.
Im not sure yet what is the reason of initial disconnection but we have such case:

1. We have already 32 routers in our 140 nodes network, routers are not downgrading possibly due to not having 7 neighbors with (link quality 2)
2. Device sends a Parent Request for some reason (already in different partiton, lost his parent, lost a leader?)
3. Initial parent request asks only routers for an answer
4. Node is collecting parent responses and none of the routers has link quality 3
5. Node sends second parent request to routers and REEDs
6. We have lots of answers and the best match (link 3) is some child
7. We ask that child to be our parent but it fails with ADDR_SOL because of max routers in the network
8. We get no answer for Child ID Request
9. The same path can be replied forever and node is always trying to connect to the same child (creating own partition in the meanwhile?)
10. Eventually node will find some link 3 router or would not receive a parent response from a link 3 child successfully attaching to some router.

This issue is causing very long disconnections and big radio traffic due to Parent Responses (even 25% of the whole traffic).
Im planning to patch OT allowing to attach to a router with link quality < 3 when there are already 32 routers in the network. Its better for us to have link 2 parent than no parent.
Additionally Im thinking about decreasing kMinDowngradeNeighbors from 7 to lets say 3.
Please let me know what you think.

simonlingoogle, 2020-04-18:
> Additionally Im thinking about decreasing kMinDowngradeNeighbors from 7 to lets say 3.

We have observed a similar situation in [OTNS](https://github.com/openthread/ot-ns) simulation that two or more partitions can not merge because there are already 32 routers in the major partition and routers do not downgrade for a very long duration causing significant traffic. @librasungirl 

The script for simulating partition merging is [form_partition.py](https://github.com/openthread/ot-ns/blob/master/pylibs/examples/form_partition.py).

I simulated 8x8 nodes for 10 times, and got following result:

- 8x8 nodes, kMinDowngradeNeighbors=7: 

| Simulation | Duration (s) | Partition Num |
|------------|--------------|---------------|
| 1#         |   1800         |   4           |
| 2#         |     **20105**       |         7      |
| 3#         |      27        |       0        |
| 4#         |      1610       |       5        |
| 5#         |      19        |       0        |
| 6#         |      263 |       3        |
| 7#         |      607 |       5       |
| 8#         |      **7533** |       9        |
| 9#         |      **>10 hours** |       take too long, seems never end |
| 10#         |      604 |       3        |

* The first value of each column is the duration it takes for 64 nodes to form one partition. 
* The second value of each column is the number of different partitions that showed up in the simulation after the initial 30 seconds. **This value can be an indication of the stability of the network**. Partitions in the initial 30 seconds are ignored because lots of nodes become Leader at the same time at the beginning of simulation. 

**There are 3 cases that took hours before merging to one partition.** In these cases, the nodes at the boundary of the major partition (the partiton containing most nodes) are all REEDs that unable to become Routers. 

I changed `kMinDowngradeNeighbors` to 3 and simulated again:

- 8x8 nodes, kMinDowngradeNeighbors=3:

| Simulation | Duration (s) | Partition Num |
|------------|--------------|---------------|
| 1#         |    539       |         26     |
| 2#         |    480       |     17          |
| 3#         |    158         |     6          |
| 4#         |       310     |       7        |
| 5#         |    2606         |       116        |
| 6#         |   510   |       35        |
| 7#         |  893    |  48      |
| 8#         |  254    |       10       |
| 9#         |  649  |    49    |
| 10#         |  531   |  44     |

My intuitional feeling is that the major partition is much less stable with much more partitions appeared during the simulation. However, there is no case that take extreme long time to form one partition like case 2#, 8#, 9# of `kMinDowngradeNeighbors == 7`. The data above also confirmed that. 

I would suggest us analyze more on the impact of different `kMinDowngradeNeighbors` values (maybe using [OTNS](https://github.com/openthread/ot-ns)) and explore other possible solutions.
@jwhui @mwilniewiec

simonlingoogle, 2020-04-18:
I got coredumps on macOS when simulating 8x8 nodes with `kMinDowngradeNeighbors == 4` and `OPENTHREAD_CONFIG_MLE_LONG_ROUTES_ENABLE=1`
```
* thread #1, stop reason = signal SIGSTOP
  * frame #0: 0x00007fff5ecf82c2 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff5edb3bf1 libsystem_pthread.dylib`pthread_kill + 284
    frame #2: 0x00007fff5ec626a6 libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff5ec2b20d libsystem_c.dylib`__assert_rtn + 324
    frame #4: 0x0000000108bc9fc8 ot-cli-ftd`ot::Mle::MleRouter::HandleParentRequest(this=<unavailable>, aMessage=<unavailable>, aMessageInfo=<unavailable>) at mle_router.cpp:1622:5 [opt]
    frame #5: 0x0000000108bc4221 ot-cli-ftd`ot::Mle::Mle::HandleUdpReceive(this=0x0000000108c19cc8, aMessage=0x0000000108c16790, aMessageInfo=0x00007ffee7086b90) at mle.cpp:2889:26 [opt]
    frame #6: 0x0000000108bb3c49 ot-cli-ftd`ot::Ip6::Udp::HandleMessage(this=0x0000000108c18ff0, aMessage=0x0000000108c16790, aMessageInfo=0x00007ffee7086b90) at udp6.cpp:336:5 [opt]
    frame #7: 0x0000000108bae163 ot-cli-ftd`ot::Ip6::Ip6::HandleDatagram(ot::Message&, ot::Ip6::Netif*, void const*, bool) [inlined] ot::Ip6::Ip6::HandlePayload(this=0x0000000108c18f98, aMessage=<unavailable>, aMessageInfo=0x00000000000002ff, aIpProto='\x11') at ip6.cpp:1002:22 [opt]
    frame #8: 0x0000000108bae13d ot-cli-ftd`ot::Ip6::Ip6::HandleDatagram(this=0x0000000108c18f98, aMessage=0x0000000108c16790, aNetif=0x0000000108c19108, aLinkMessageInfo=<unavailable>, aFromNcpHost=false) at ip6.cpp:1253 [opt]
    frame #9: 0x0000000108bbd4cd ot-cli-ftd`ot::MeshForwarder::HandleLowpanHC(unsigned char const*, unsigned short, ot::Mac::Address const&, ot::Mac::Address const&, otThreadLinkInfo const&) [inlined] ot::MeshForwarder::HandleDatagram(this=<unavailable>, aMessage=<unavailable>, aLinkInfo=0x00007ffee7086cb0, aMacSource=<unavailable>) at mesh_forwarder.cpp:1337:28 [opt]
    frame #10: 0x0000000108bbd4a0 ot-cli-ftd`ot::MeshForwarder::HandleLowpanHC(this=0x0000000108c19b70, aFrame="\x7f;\x02?MLML?3", aFrameLength=<unavailable>, aMacSource=0x00007ffee7086ca0, aMacDest=0x00007ffee7086c90, aLinkInfo=0x00007ffee7086cb0) at mesh_forwarder.cpp:1311 [opt]
    frame #11: 0x0000000108bbd0d7 ot-cli-ftd`ot::MeshForwarder::HandleReceivedFrame(this=<unavailable>, aFrame=<unavailable>) at mesh_forwarder.cpp:1041:13 [opt]
    frame #12: 0x0000000108b99655 ot-cli-ftd`ot::Mac::Mac::HandleReceivedFrame(this=0x0000000108c198c8, aFrame=<unavailable>, aError=<unavailable>) at mac.cpp:1801:26 [opt]
    frame #13: 0x0000000108bdd7dc ot-cli-ftd`platformRadioReceive [inlined] radioProcessFrame(aInstance=<unavailable>) at radio.c:770:13 [opt]
    frame #14: 0x0000000108bdd79e ot-cli-ftd`platformRadioReceive [inlined] radioReceive(aInstance=0x0000000108c10320) at radio.c:500 [opt]
    frame #15: 0x0000000108bdd6b9 ot-cli-ftd`platformRadioReceive(aInstance=<unavailable>, aBuf=<unavailable>, aBufLength=<unavailable>) at radio.c:596 [opt]
    frame #16: 0x0000000108bded78 ot-cli-ftd`otSysProcessDrivers [inlined] receiveEvent(aInstance=<unavailable>) at platform-sim.c:111:9 [opt]
    frame #17: 0x0000000108bded18 ot-cli-ftd`otSysProcessDrivers(aInstance=<unavailable>) at platform-sim.c:304 [opt]
    frame #18: 0x0000000108b78b20 ot-cli-ftd`main(argc=2, argv=0x00007ffee7087808) at main.c:114:9 [opt]
    frame #19: 0x00007fff5ebbd3d5 libdyld.dylib`start + 1
    frame #20: 0x00007fff5ebbd3d5 libdyld.dylib`start + 1
```
This simulation never coredumped without `LONG_ROUTES`. So, I would expect it has something to do with the `LONG_ROUTES` extension. 
Will investigate later.

jwhui, 2020-04-18:
@simonlingoogle, thanks for sharing your findings. 

@mwilniewiec, can you confirm again what your topology is like? Your [original post](https://github.com/openthread/openthread/issues/4508#issue-554924331) indicated a long linear topology where devices only had 3-4 neighbors. In that case, I can certainly see why `kMinDowngradeNeighbors = 7` is too large. But with 140 nodes, maybe the density is higher. How many neighbors per device and total hops do you expect in your topology?

mwilniewiec, 2020-04-18:
We are using a simulation setup, trying to keep RSSI of closest nodes ~75dbm. We did tests with different topologies with different number of devices. We started from 50 linear but as we were growing to 140 it was a bit hard to keep them linear as it took a lot of space.
Currently Im doing tests on 140 hardware devices where currently topology is more or less something like:
```
NNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNN
NNNNNNNNNNNNNNN       BR NNNNNNNNNNNNNNNNNNNNNNNNNN
                            NNNNNNNNNNNNNNNNNNNNNN
                              NNNNNNNNNNNNNNNNNNNNN
```
N is a node and BR is a border router with NCP. We track their status from the cloud. When device is in a different partition than NCP it is disconnected for us.

Example neighbor table and scan results from 2 different, random nodes:
```
> neighbor table
| Role | RLOC16 | Age | Avg RSSI | Last RSSI |R|S|D|N| Extended MAC     |
+------+--------+-----+----------+-----------+-+-+-+-+------------------+
|   R  | 0x1800 |   0 |      -91 |       -93 |1|0|1|1| da9c8f0f52b3b369 |
|   R  | 0x5c00 |   3 |      -77 |       -78 |1|0|1|1| 8e79af0446e57223 |
|   R  | 0x9000 |   3 |      -75 |       -75 |1|0|1|1| eef4fbd8888d4d91 |
|   R  | 0xec00 |   1 |      -67 |       -67 |1|0|1|1| fe4c44b7a4cb5c28 |
```
```
> scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
> | 1 | biotcloud-40     | 3c9b17163421fa72 | 6acb | 0a07c52fdc1b10f4 | 22 | -82 |  48 |
| 1 | biotcloud-40     | 3c9b17163421fa72 | 6acb | 021ac9f6b665dca2 | 22 | -70 |  92 |
| 1 | biotcloud-40     | 3c9b17163421fa72 | 6acb | ea0680f5124bb38e | 22 | -87 |  20 |
| 1 | biotcloud-40     | 3c9b17163421fa72 | 6acb | 1abe48e43c442031 | 22 | -84 |  32 |
| 1 | biotcloud-40     | 3c9b17163421fa72 | 6acb | fe4c44b7a4cb5c28 | 22 | -92 |   4 |
| 1 | biotcloud-40     | 3c9b17163421fa72 | 6acb | be25a412dc0fcf91 | 22 | -87 |  20 |
```
In general I see that devices link with 2-4 neighbors in every direction.
Ive also seen cases where node was a child of a parent 10 devices away from it with rssi 90

mwilniewiec, 2020-05-15:
Hi again,
I have some more news and more questions to ask.
I created a testing machinery for 140 nodes farm, did some improvements and before and after tests.
My general idea was to sniff packets that appear too often, understand and reduce them + increase default radio retries for better overall communication.
Changes that were done:
1. Changed OPENTHREAD_CONFIG_MAC_MAX_FRAME_RETRIES_DIRECT from 3 to 8
2. integrated #3358 and set OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS = 1
3. intergrated #4753 
4. Changed HandleParentResponse passing through a linkQuality < 3 router when it has kMaxRouters in connectivity TLV
5. Changed kMinDowngradeNeighbors from 7 to 4
6. Limited and randomized application packets
7. Added COAP re-transmissions on border router [br #396](https://github.com/openthread/ot-br-posix/issues/396)

With those changes I can see much less packets sent over network, partitions are much more stable with no nodes leaving partitions for longer period of time (seen before adding point 4).
General packets delivery BR -> nodes is around 3 times better and reliable than before.

There is still room for improvement and I have some ideas like:
1. Stop using border router ML-EID address (must transfer RLOC packets from ncp->linux which can be tricky) + nodes will address border router by its service ALOC
2. Go higher with radio retries, @jwhui suggested 15
3. Upfront precaching of EID-RLOC entries on NCP not to flood network with a/aq multicasts when suddenly a command is sent to multiple or all nodes.

My newest finding is that I see lot of Link Requests from multiple routers to a router sending an advertisement. 13% of all traffic now are Link Requests (10 000) with almost no Link Accepts (70).
Any thoughts why this could happen?

jwhui, 2020-05-15:
> With those changes I can see much less packets sent over network, partitions are much more stable with no nodes leaving partitions for longer period of time (seen before adding point 4).
> General packets delivery BR -> nodes is around 3 times better and reliable than before.

@mwilniewiec,  great to hear that your network performance is improving!

> There is still room for improvement and I have some ideas like:
> 
> 1. Stop using border router ML-EID address (must transfer RLOC packets from ncp->linux which can be tricky) + nodes will address border router by its service ALOC

Makes sense. This would avoid the EID-to-RLOC lookup.

The issue with RLOC addresses becomes much simpler when using the RCP model (which puts OpenThread on the host side). ot-br-posix master now defaults to RCP mode and we have updated the Border Router guide on openthread.io to match.

> 2. Go higher with radio retries, @jwhui suggested 15

This depends on the loss rates you are experiencing.

> 3. Upfront precaching of EID-RLOC entries on NCP not to flood network with a/aq multicasts when suddenly a command is sent to multiple or all nodes.

Last year, we added the ability to update the EID-to-RLOC cache upon receiving a message (#4091). This allows devices to update a receiver's cache simply by sending a message.

> My newest finding is that I see lot of Link Requests from multiple routers to a router sending an advertisement. 13% of all traffic now are Link Requests (10 000) with almost no Link Accepts (70).
> Any thoughts why this could happen?

I wonder if this is due to asymmetric link qualities. Are the Link Requests mostly coming from active routers, router-eligible end devices (REEDs), or a mix? Looking at the code, active routers only send Link Requests if the received Advertisement is above some RSSI threshold. However, REEDs do not apply the same - we could look into apply the same RSSI filter if we think this will improve things.

mwilniewiec, 2020-05-17:
@jwhui there were lot of Link requests from REEDs before I applied point 2 from the list.
Now its only active routers that send link requests from what I see. I also dont know why I can't see radio ACKs for the link requests in the sniff. Wireshark screen attached.
[image]

mwilniewiec, 2020-06-02:
@jwhui I tried to investigate those strange Link Requests without a response a bit more.
From what I see that are a few routers in my setup that are getting those link requests on every Advertisement (mostly every 5 sec) and not responding to it (also radio ACKS are missing). This is causing quite a lot of unnecessary traffic.
From the application perspective those devices work normally as expected (sending application packets properly, acking them). I also checked commands from CLI, bufferinfo, router table, everything looked fine. Routers that are requesting for a link are in the router table but are not in the neighbor table. Neighbor table has some other entries.

In the same sniff I also see other routers that work as expected. Sending Adverisement, responding to a Link request (if they get any) with a radio ACK and Link Accept.
All devices have the same software.

I also tried to reboot such a router to see if its persistent but after a reboot it acted a bit strage ...
Within 30 ms sent:
1. Link request to ff02:2
2. Two parent reqests (one only for routers, the other one for REEDs)
3. Data Response  to ff02:1 (with a new partition data)
4. Advertisement (saying its a new leader)
After that it started receiving Parent Responses, picked one parent and became a child

Do you know any fixes that were done in these areas?
Any idea what could be the reason of a router not responding to link requests?

jwhui, 2020-06-02:
Thanks for continuing to investigate and keeping us up to date on your findings!

> @jwhui I tried to investigate those strange Link Requests without a response a bit more.
> From what I see that are a few routers in my setup that are getting those link requests on every [...]

By "getting those link requests" can you tell that the device is actually receiving those (e.g. via logs)?

> From the application perspective those devices work normally as expected (sending application packets properly, acking them). I also checked commands from CLI, bufferinfo, router table, everything [...]

In the case of A sending a Link Request to B. Can you share whether A is also in B's router table? A and B should have each other in their router table.

> In the same sniff I also see other routers that work as expected. Sending Adverisement, responding to a Link request (if they get any) with a radio ACK and Link Accept.
> All devices have the same [...]

In the above example, are you rebooting A (the sender of Link Requests not receiving ACKs) or B (the device not responding to Link Requests) or both?

> Within 30 ms sent:
>
> 1. Link request to ff02:2
> 2. Two parent reqests (one only for routers, the other one for REEDs)
> 3. Data Response  to ff02:1 (with a new partition data)
> 4. Advertisement (saying [...]

The fact that Parent Responses are received only after becoming a Leader and sending Advertisement messages is not normal.

1. Were there any additional Parent Request messages sent between steps 2 and 4 above?

1. Did the Parent Response message have a Response TLV value matching the Challenge TLV value in the Parent Request from step 2?

1. What are the time delays between each of the messages? I'm wondering if the neighboring device is taking too long to respond to the request for some reason.

If possible, could you share the pcap file that demonstrates the above behavior?

> Do you know any fixes that were done in these areas?
> Any idea what could be the reason of a router not responding to link requests?

Unfortunately, nothing comes to the top of my mind. I think having logging information could help determine whether neighboring devices are receiving the Link Request and, if so, whether they are trying to send Link Accept messages in response.

mwilniewiec, 2020-06-03:
I made more investigation today and I think I know where the problem with Link Requests lies.
As you said "asymmetric link qualities" can be an issue.
My setup is using different antennas and it seems that:
1. Router A is sendng Advertisement
2. Routers R1, R2, R2 are receiving advertisement with link margin > 10
3. Routers R1, R2, R3 are sending Link Request to A (with OPENTHREAD_CONFIG_MAC_MAX_FRAME_RETRIES_DIRECT retries)
4. Router A can't hear those requests and isn't sending even a radio ACK
5. Such traffic persists on every Advertisement sent by A

I tried to use a better antenna for any of R1, R2, R3 and suddenly A could hear a Link Request, responding with "Link Accept and Request". I'm not sure how this issue would look like in a real environment where antennas are the same for all devices.

I have a plan to introduce an improvement:
"Router A is not responding to Link Requests, do not send another Link Request for 5 minutes for that router (possibly with increasing backoff)" 
This should reduce unnecessary Link requests from ~20% to almost 0. What do you think about it?

PS. When I look at one longer sniff with low application layer traffic it is showing:

- 47% - MLE
   - 27% - Link Request
   - 10 % - Advertisement
   - 5% - Parent response
- 25% - Radio ACK
- 21% - Coap (mostly application ~19%)
- 5% Beacon

It looks like this Link Request is currently a biggest unnecessary traffic here.

According to the second topic:
> 1. Were there any additional Parent Request messages sent between steps 2 and 4 above?

No, 2 parent request one after another (Time: 28.098 & 28.104), more parent request a bit later after parent responses (28.70 for routers only and 29.45 for REEDs) 

>2 .Did the Parent Response message have a Response TLV value matching the Challenge TLV value in the Parent Request from step 2?

Yes, both parent requests have corresponding responses (matching Response TLV)

> 3. What are the time delays between each of the messages? I'm wondering if the neighboring device is taking too long to respond to the request for some reason.

Attached screen from the sniff, I can send full sniffs & keys over private channel if you'd like to look deeper. It can be interesting as there are 140 devices.
[image]

LuDuda, 2020-06-06:
@mwilniewiec regarding the timings, I wonder if you can get logs with timestamps from the device which sends MLE Link Request, Parent Requests and Data Request right after the other without respecting the timings. One of the causes could be that there is a problem with alarm implementation, but I wonder if it can be caused by the application itself. Is it possible that the main loop is not executed constantly in your case but it is blocked for some time? That would explain why timers fired in this short time.

Some more questions..
1) Did you modify the CSMA/CA parameters?
2) Do you use any operating system e.g. FreeRTOS?
3) Can you see this timing problem on any boards, or it may be related to the particular hardware board?

mwilniewiec, 2020-07-01:
@LuDuda thanks, good catch with blocked main task. I found that our code called otThreadSetEnabled(true), blocked from 0 to 4 seconds and started feeding thread in regular loop. It looks like this was the root-cause of strage packets during wakeup.

To answer your question:

1. ```Did you modify the CSMA/CA parameters?``` Currently we are using NRF_RADIO_CCA_MODE_CARRIER_AND_ED instead of NRF_RADIO_CCA_MODE_ED (planning to get back with some regression testing), we changed NRF_802154_CCA_ED_THRESHOLD_DEFAULT and NRF_802154_CCA_CORR_THRESHOLD_DEFAULT both from 20 to 10
2. ```Do you use any operating system e.g. FreeRTOS?```, Yes we are using FreeRTOS

mwilniewiec, 2020-07-01:
I would like to summarize where we are now with our big network testing.
We are still testing 140 Nodes with 1 NCP, we try to keep RSSI between nearest neighbors ~75dBm. Our test process is as follows:

1. All nodes send application ping to NCPs Mesh Local Address every 2 minutes (using their MeshLocal as a source address). This way NCP can have 140 entries in his EID-RLOC Cache (#4091 applied)
2. Border Router sends Confirmable COAP to all nodes Mesh Local addresses in 200ms intervals and passes the response to the cloud. Nodes response is quite big and fragmented into 3 frames. 
3. Request is retried for all nodes that did not respond with increasing backoff staring from 30 sec.
4. Such test (sending requests to all nodes) is run multiple times every 10 minutes.

Improvements that were done, mainly reducing overal unnecessary traffic in order to have more room for application message retransmissions:

1. Changed OPENTHREAD_CONFIG_MAC_MAX_FRAME_RETRIES_DIRECT from 3 to 10
2. integrated #3358 and set OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS = 1
3. intergrated #4753
4. Changed HandleParentResponse passing through a linkQuality < 3 router when it has kMaxRouters in connectivity TLV
5. Changed kMinDowngradeNeighbors from 7 to 4
6. Limited and randomized application packets
7. Added COAP re-transmissions on border router https://github.com/openthread/ot-br-posix/issues/396
8. Applied #4091 so that nodes and NCP shouldn't need to use a/aq multicast at all
9. Added backoff (up to 5 min) for link request to nodes that did not respond previously, see https://github.com/openthread/openthread/issues/4508#issuecomment-638268332

After all those improvements sending req-resp to all nodes works much better. 

Previously we had around 40 from 140 responses in 10 minutes and many tests with 0 responses as NCP become it's own leader in a different partition unable to merge because of point 4 from the improvements list. We could also see that application pings were lost (node not seeing NCP).

Now we have mostly 140/140 responses in around 5 minutes with average response time ~2 min. When network is "idle" with only thread traffic + application pings everything works flawlessly, during req-resp tests nodes sometimes lose their connection for a few miniutes and comes back.

Some tests show us that we are able to get all 140 responses in 2 minutes and some fail with only 80 of 140 nodes responding in 10 minutes with many nodes losing connection to NCP.
I also did some testing with multiple regular antennas and part of the nodes & NCP seeing one another as neighbours. Such test works great with immediate 140 answers ~ 1 minute.

Im trying to understand where the problem lies when results are bad and what is different between tests. For now Im not assuming that radio environment can be so different in our test setup.
Im still observing that nodes are changing their RLOCs a lot during testing (every 10 minutes we have different RLOCs on several nodes) and this is not helping for sure. @jwhui any hints what to check, how to investigate RLOC changes?

Future plans:
1. Use muticast to Realm local as a first request with randomly distributed responses to reduce traffic.
2. Investigate RLOC changes and network partitioning during req-resp tests.
3. Investigate why we observe huge differences in results on the same software.
4. Put Thread NCP on border router host according to new OT architecture and recommendations.

konradderda, 2020-07-02:
@mwilniewiec Just one note, you mentioned that: 

>     1. `Did you modify the CSMA/CA parameters?` Currently we are using NRF_RADIO_CCA_MODE_CARRIER_AND_ED instead of NRF_RADIO_CCA_MODE_ED (planning to get back with some regression testing), we changed NRF_802154_CCA_ED_THRESHOLD_DEFAULT and NRF_802154_CCA_CORR_THRESHOLD_DEFAULT both from 20 to 10

Changing the default value from 20 to 10 changes the CCA threshold from -72 dBm to -82 dBm which can significantly reduce the device's transmission opportunity (every ED result above -82 dBm blocks the transmission). Please ensure that it's expected behavior.

mwilniewiec, 2020-07-03:
@konradderda our RSSI between devices in more 80tish than 70tish. If I understand that correctly having CCA enabled from -72dBm threshold would almost disable it totally. I believe that our setups can be noisy and it seems resonable to block transmission instead of of filling the ether fully. Still thanks for the note, we should actually compare both values. 
PS. Counters in our test setup shows that blocking transmission due to CCA isn't a big issue:
```
TxTotal: 17617
    TxUnicast: 3900
    TxBroadcast: 13717
    TxAckRequested: 3709
    TxAcked: 3441
    TxNoAckRequested: 13908
    TxData: 17235
    TxDataPoll: 0
    TxBeacon: 191
    TxBeaconRequest: 191
    TxOther: 0
    TxRetry: 4176
    TxErrCca: 302
    TxErrBusyChannel: 8
```

jwhui, 2020-11-04:
@mwilniewiec , do you have any updates on this issue?

mwilniewiec, 2020-11-05:
Lately we didn't do much testing as we achieved let's say "satisfying performance" and focused on different issues. One thing that I mentioned earlier is that we did a few tests that went almost perfect but lately its getting only worse and worse (even tests on the same software were showing generally worse results). 
I think that we also recently added some bigger regression (for example we added a mutex for all OT API calls, this could be a reason) and  we are getting many disconnections during tests.
Additionally we successfully merged Nordic SDK 4.1 with newer OT and we are ready to make tests on this SW.
I'm planning to do more investigations very soon, why do we see a regression + test on SDK 4.1 if it improves anything and fixes bugs that we are still experiencing.
If it's OK for you I would keep the issue opened and we should all try to achieve better responsiveness in huge > 100 nodes OT networks. In the past we were expecting a lot more from OT and hardware setup as still we are struggling with the fact that lots of packets are lost + OT network breaks apart and needs to recover when there ale lots of packets moving around.

jwhui, 2020-11-06:
@mwilniewiec , thanks for the update. We should work together on improving performance whenever we can.

mwilniewiec, 2020-12-21:
@jwhui another updade from my side. I did more investigation on frequently changing RLOCS and I did some improvements on OT constants.
From what Ive seen from logs and sniffs the flow was as folows:
1. There was a lot of traffic in our network due to sending messages
2. Child is waiting 4 minutes - 4 seconds to send Child Update Request to his parent with 3 retries (every 1 second)
3. While sending it fails on a NO_ACK because the ether is noisy. Sometimes it fails even after 1 or 0 retries because previously it sent other packet to the parent with NO_ACK.

I changed:
```
-    kMaxChildKeepAliveAttempts = 4, ///< Maximum keep alive attempts before attempting to reattach to a new Parent
+    kMaxChildKeepAliveAttempts = 6, ///< Maximum keep alive attempts before attempting to reattach to a new Parent
 
-    kUnicastRetransmissionDelay     = 1000,  ///< Base delay before retransmitting an MLE unicast.
+    kUnicastRetransmissionDelay     = 2000,  ///< Base delay before retransmitting an MLE unicast.

-    kMaxTransmissionCount           = 3,     ///< Maximum number of times an MLE message may be transmitted.
+    kMaxTransmissionCount           = 5,     ///< Maximum number of times an MLE message may be transmitted.

-    kFailedRouterTransmissions = 4,   ///< FAILED_ROUTER_TRANSMISSIONS
+    kFailedRouterTransmissions = 10,   ///< FAILED_ROUTER_TRANSMISSIONS
```
Where ```kFailedRouterTransmissions``` is the most valuable from what I understand.
Can you see any risk related with applying such configuration?

With those settings Network works better, isn't disconnecting so often.
Previously we had a snowball effect:
1. Lot of application traffic caused child to detach from parent
2. Detach caused lots of parent request/response
3. Detach caused eidcache being lost and needed a/aq a/an to get eid mapping for border router
4. RLOC change caused another a/aq a/an to get EID->RLOC for that node on our NCP 
6. Too many packets and detached state caused app packets being lost and retried making more packets in the network
As a result we had a snowball where OT packets are fighting with application packets to reach the target.

This is a step forward but still more to be done. We are waiting for Nordic to analyse SDK 4.1 upgradability problem and we want to switch on that. Regards & a happy new year :)

mwilniewiec, 2022-03-14:
Closing as I have no further progress for now. If I do so Ill open new issues with more detail.

Summary: Tested **140 hardware devices** in a single network + 1 border router.
When devices send packets **hourly** or **once every several minutes** everything looks stable and works fine.
When we do stuff like sending **many messages** to different ML-EIDs at once or multicast with **lots of responses** or device messages are not evenly distributed and all want to send packet exactly 10 sec after startup things can go wrong.
I would expect OT topology to be a bit more stable as I see that RLOCs change too often and when everything reorganizes auto-healing can take a few minutes or more. 
Still ... when it works - it works :)

closed 2022-03-14 by mwilniewiec

simonlingoogle, 2022-03-15:
Obviously we have some valuable configuration changes that work better in large-scale networks as in https://github.com/openthread/openthread/issues/4508#issuecomment-748853553

How do we adopt these changes to OT code that other may reuse them (without having to figure them out again) ?
Maybe we can add a document for recommended OT configurations in large-scale networks?

Thoughts? @mwilniewiec @jwhui

mwilniewiec, 2022-03-17:
I pointed out multiple other improvements that I've done here:
https://github.com/openthread/openthread/issues/4508#issuecomment-652328664

Some changes from the list were applied only as local patches and never contributed to OT. Its hard to say if its still valid but it was helpful to reduce unnecessary traffic & get satisfying network performance.
For example:
4: https://github.com/openthread/openthread/issues/4508#issuecomment-615873005
9: https://github.com/openthread/openthread/issues/4508#issuecomment-638268332

Answering to your question I think that some configuration + good practices for large-scale networks would be helpful because people wanting to use it this way have no idea where to start. It also depends if it's your target at all.
In general Thread is presented as:
> best way to connect and control products in the home and buildings

But ...We decided to use it for city infrastructure where number of devices and distances between them drastically change.
And... it works :)

weiqj, 2022-05-13:
> @mwilniewiec Just one note, you mentioned that:
>
>
> Changing the default value from 20 to 10 changes the CCA threshold from -72 dBm to -82 dBm which can significantly reduce the device's transmission [...]

Hello I would like to point out that, according to NRF documentation, CCA register to dbm conversion is:
PRF[dBm] = ED_RSSIOFFS + ED_RSSISCALE x VALHARDWARE
For NRF52840, ED_RSSIOFFS = -92, ED_RSSISCALE=4
Value of 20 results in -12dBm, value 10 results in -52dBm.

Please correct me if I was wrong.