openthread/openthread #4512: Site survey CLI command

cfaessler, 2020-01-28:
**Is your feature request related to a problem? Please describe.**
For a new home automation product line I am evaluating OpenThread technology. One of the key performance metrics for the evaluation is possible coverage with meshed devices. With the current solution it is hard to evaluate different scenarios in a semi-automated way.

**Describe the solution you'd like**
Client side: Add a new CLI command which receives UDP broadcasts and replies to a configured server with link quality, rssi and node name values.
To send broadcasts and log received values I would provide a python script. Mainly intended to run on the border router.

**Describe alternatives you've considered**
- Alternatively to the python script, one could implement a CLI udp server part
- Future idea could be to integrate this tool in the border router UI, which let's one simple do a node exploration / site survey

cfaessler, 2020-02-05:
@jwhui does no labeling and reaction mean there's no interest on this?

bukepo, 2020-02-05:
@cfaessler Thank you for proposing this feature. We have a new tool in https://github.com/openthread/openthread/pull/4117, which seems to be able to serve as a base of this feature. It's close for now, but we are going to re-open it later.

Could you please help review it and see if it's helpful to your use case?

jwhui, 2020-03-18:
@cfaessler, any updates you can share?

jwhui, 2020-04-16:
Closing stale issue.

closed 2020-04-16 by jwhui